thesofproject/sof #9888: [BUG]Silence at the beginning of playback/capture with IPC4

ranj063, 2025-03-10:
**Describe the bug**
There's always a minimum of 4 periods of silence at the beginning of playback/capture with IPC4.
And the silence increases with increased DAI DMA buffer sizing.

This is because we trigger the sink pipelines and therefore the link DMA first. So, the first data comes out after the read pointer wraps around the entire buffer (which is a minimum of 4 periods long).
Note: See the screenshot of the waveforms below.  The top waveform is with the DMA buffer size of 16 periods, the second one with 8 periods and the last one is the default with 4 periods.

**To Reproduce**
Any aplay command ex: aplay -D:hw0,0 file.wav

**Reproduction Rate**
100%

**Expected behavior**
No silence at the beginning or maybe minimal silence

**Impact**
Not a showstopper

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: topic/sof-dev
    * SOF: main
2) Name of the topology file
    * Topology: sof-ptl-nocodec.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: PTL

[image]

lgirdwood, 2025-03-10:
Solution needs to be we add Kconfig to the platform min period count. @lyakh do you have time to add Kconfigs for the min num periods in link/host DMA drivers, IIRC these values were picked due to safety but are non optimal. 

https://github.com/thesofproject/sof/blob/c00f067939233c28a5e516941a821b032ef81879/zephyr/lib/dma.c#L88

ranj063, 2025-03-11:
> Solution needs to be we add Kconfig to the platform min period count. [@lyakh](https://github.com/lyakh) do you have time to add Kconfigs for the min num periods in link/host DMA drivers, IIRC these values were picked due to safety but are non optimal.
> 
> [sof/zephyr/lib/dma.c](https://github.com/thesofproject/sof/blob/c00f067939233c28a5e516941a821b032ef81879/zephyr/lib/dma.c#L88)
> 
> Line 88 in [c00f067](/thesofproject/sof/commit/c00f067939233c28a5e516941a821b032ef81879)
> 
>  .period_count	= HDA_DMA_BUFFER_PERIOD_COUNT,

@lgirdwood even if we add a min period count, that will determine the number of samples of silence at the beginning isnt it?

lgirdwood, 2025-03-11:
> [@lgirdwood](https://github.com/lgirdwood) even if we add a min period count, that will determine the number of samples of silence at the beginning isnt it?

@ranj063 if Kconfig sets it to Ping/pong then we guarantee a 1 period delay. Further work can start the link DMA on first data

lgirdwood, 2025-03-11:
@lyakh this can be done as background as you proceed thru your other items. Just need a Kconfig for the link DMA period counts instead of the hard coded macro value. Pls set default Kconfig to 2 periods.

ranj063, 2025-03-11:
> [@lyakh](https://github.com/lyakh) this can be done as background as you proceed thru your other items. Just need a Kconfig for the link DMA period counts instead of the hard coded macro value. Pls set default Kconfig to 2 periods.

@lgirdwood I can also do this if @lyakh is busy

lyakh, 2025-03-12:
> [@lgirdwood](https://github.com/lgirdwood) I can also do this if [@lyakh](https://github.com/lyakh) is busy

I think this should also be tested? ATM I have no idea how to do that - capture per loopback? So if @ranj063 knows how to do that (since she's already done that for this report) and has all the instrumentation set up, maybe it's easier for her also to make this patch? Works either way for me. And yes, thanks for offering to take this over, appreciated!

ranj063, 2025-03-12:
> I think this should also be tested? ATM I have no idea how to do that - capture per loopback? So if [@ranj063](https://github.com/ranj063) knows how to do that (since she's already done that for [...]

loopback won't help because there will be a delay in starting arecord after starting aplay. I'll do the measurements with our set up.

wszypelt, 2025-04-25:
@ranj063 any update?

ranj063, 2025-05-22:
@wszypelt Im sorry I havent had the chance to look into this yet.

closed 2025-08-08 by github-actions[bot]