thesofproject/sof #9891: [BUG]

khpylon, 2025-03-11:
**Describe the bug**
When hibernating on a Lenovo laptop attached to an external HDMI monitor, the system will be take an extra 30 to 45 seconds to power off, and upon resuming take an extra 30 seconds before becoming responsive (i.e., the video displays are restored but cannot be interacted with). 

**To Reproduce**
Attach an external HDMI monitor to the laptop, and put into hibernation.

**Reproduction Rate**
This will occur every time hibernation is initiated.  

**Expected behavior**
The laptop should power off within approximately 15 seconds, and upon resume the UI should be responsive as soon as the video displays are visible.

**Impact**
This is no longer a show-stopper once I determined the issue was the sound card.   I've written a script that is run by systemd when hibernation begins.  

**Environment**
    * Kernel: Linux 6.13.5-arch1-1 #1 SMP PREEMPT_DYNAMIC Thu, 27 Feb 2025 18:09:44 +0000 x86_64 GNU/Linux
    * SOF firmware version: 2:2:0-57864
    * SOF topology: ABI 3:22:1 
    * Kernel ABI: 3:23:1
    * Firmware file: intel/sof/sof-rpl.ri
    * Toplogy file: intel/sof-tplg/sof-hda-generic-2ch.tplg
    * Distrbution: Arch
    * Hardware: Lenovo Ideapad 5i 16IRU9
    * Chipset: Intel Raptor Lake-P/U/H cAVS driver
    * ALSA version: k6.13.5-arch1-1
    * Sound server: PipeWire v: 1.2.7 

**Screenshots or console output**

[dmesg.txt](https://github.com/user-attachments/files/19192605/dmesg.txt)

lgirdwood, 2025-03-12:
@khpylon audio is indeed coupled to the display driver when a monitor is connected, but I am concerned about this warning from the display driver which could impact the audio driver and system at suspend() (since BIOS, display driver and audio driver all have work at suspend). Btw, what does your systemd script do ? Can you share as this may help root cause.
```
[    1.447448] ------------[ cut here ]------------
[    1.447455] i915 0000:00:02.0: [drm] Port A asks to use VBT vswing/preemph tables
[    1.447493] WARNING: CPU: 3 PID: 183 at drivers/gpu/drm/i915/display/intel_bios.c:2692 intel_bios_init+0x1352/0x17f0 [i915]
[    1.448006] Modules linked in: dm_mod i915(+) crct10dif_pclmul crc32_pclmul crc32c_intel polyval_clmulni polyval_generic i2c_algo_bit ghash_clmulni_intel drm_buddy sha512_ssse3 sha256_ssse3 ttm sha1_ssse3 aesni_intel serio_raw intel_gtt sdhci_pci gf128mul atkbd nvme sdhci_uhs2 libps2 vivaldi_fmap crypto_simd drm_display_helper sdhci nvme_core spi_intel_pci cqhci cryptd mmc_core nvme_auth spi_intel cec i8042 serio video wmi
[    1.448061] CPU: 3 UID: 0 PID: 183 Comm: (udev-worker) Not tainted 6.13.5-arch1-1 #1 a7601aaf9729ecd670c97714fd422c8e98fdc244
[    1.448069] Hardware name: LENOVO 83DU/LNVNB161216, BIOS P0CN20WW 10/21/2024
[    1.448073] RIP: 0010:intel_bios_init+0x1352/0x17f0 [i915]
[    1.448427] Code: 48 8b 78 08 48 8b 5f 50 48 85 db 75 03 48 8b 1f e8 e3 a9 81 e5 44 89 f1 48 89 da 48 c7 c7 60 c2 ae c0 48 89 c6 e8 0e d2 ee e4 <0f> 0b e9 bb fd ff ff 0f b7 83 60 04 00 00 83 e8 03 66 83 f8 04 76
[    1.448431] RSP: 0018:ffffa184c08a7710 EFLAGS: 00010286
[    1.448436] RAX: 0000000000000000 RBX: ffff94e6c252a360 RCX: 0000000000000027
[    1.448439] RDX: ffff94ea5f7a18c8 RSI: 0000000000000001 RDI: ffff94ea5f7a18c0
[    1.448442] RBP: ffff94e6d07c4618 R08: 0000000000000000 R09: ffffa184c08a7590
[    1.448445] R10: ffffffffa78b5408 R11: 0000000000000003 R12: 0000000000000000
[    1.448447] R13: 0000000000000001 R14: 0000000000000041 R15: ffff94e6c8113120
[    1.448450] FS:  0000786d74c0c880(0000) GS:ffff94ea5f780000(0000) knlGS:0000000000000000
[    1.448454] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[    1.448457] CR2: 00007719c0ff63b0 CR3: 0000000100272006 CR4: 0000000000f70ef0
[    1.448461] PKRU: 55555554
[    1.448463] Call Trace:
[    1.448469]  <TASK>
[    1.448471]  ? intel_bios_init+0x1352/0x17f0 [i915 9aea23f340a24d1eb4a5ba930c2533424bb0a8b0]
[    1.448805]  ? __warn.cold+0x93/0xf6
[    1.448814]  ? intel_bios_init+0x1352/0x17f0 [i915 9aea23f340a24d1eb4a5ba930c2533424bb0a8b0]
[    1.449126]  ? report_bug+0xff/0x140
[    1.449135]  ? handle_bug+0x58/0x90
[    1.449140]  ? exc_invalid_op+0x17/0x70
[    1.449145]  ? asm_exc_invalid_op+0x1a/0x20
[    1.449153]  ? intel_bios_init+0x1352/0x17f0 [i915 9aea23f340a24d1eb4a5ba930c2533424bb0a8b0]
[    1.449452]  intel_display_driver_probe_noirq+0x42/0x240 [i915 9aea23f340a24d1eb4a5ba930c2533424bb0a8b0]
[    1.449817]  i915_driver_probe+0x3ba/0xb20 [i915 9aea23f340a24d1eb4a5ba930c2533424bb0a8b0]
[    1.450059]  ? drm_privacy_screen_get+0x198/0x1c0
[    1.450065]  local_pci_probe+0x42/0x90
[    1.450073]  pci_device_probe+0xdd/0x270
[    1.450080]  really_probe+0xdb/0x340
[    1.450085]  ? pm_runtime_barrier+0x54/0x90
[    1.450089]  ? __pfx___driver_attach+0x10/0x10
[    1.450093]  __driver_probe_device+0x78/0x110
[    1.450097]  driver_probe_device+0x1f/0xa0
[    1.450102]  __driver_attach+0xba/0x1c0
[    1.450106]  bus_for_each_dev+0x8c/0xe0
[    1.450113]  bus_add_driver+0x112/0x1f0
[    1.450116]  driver_register+0x72/0xd0
[    1.450123]  i915_init+0x23/0x90 [i915 9aea23f340a24d1eb4a5ba930c2533424bb0a8b0]
[    1.450343]  ? __pfx_i915_init+0x10/0x10 [i915 9aea23f340a24d1eb4a5ba930c2533424bb0a8b0]
[    1.450532]  do_one_initcall+0x58/0x310
[    1.450541]  do_init_module+0x60/0x230
[    1.450547]  init_module_from_file+0x89/0xe0
[    1.450555]  idempotent_init_module+0x115/0x310
[    1.450561]  __x64_sys_finit_module+0x65/0xc0
[    1.450566]  do_syscall_64+0x82/0x190
[    1.450571]  ? vfs_read+0x15e/0x370
[    1.450577]  ? vfs_read+0x15e/0x370
[    1.450582]  ? __x64_sys_pread64+0x9f/0xc0
[    1.450587]  ? syscall_exit_to_user_mode+0x37/0x1c0
[    1.450592]  ? do_syscall_64+0x8e/0x190
[    1.450595]  ? do_filp_open+0xd8/0x190
[    1.450605]  ? __pfx_page_put_link+0x10/0x10
[    1.450611]  ? do_sys_openat2+0x9c/0xe0
[    1.450615]  ? syscall_exit_to_user_mode+0x37/0x1c0
[    1.450618]  ? do_syscall_64+0x8e/0x190
[    1.450621]  ? do_syscall_64+0x8e/0x190
[    1.450624]  ? __irq_exit_rcu+0x4c/0xe0
[    1.450628]  entry_SYSCALL_64_after_hwframe+0x76/0x7e
[    1.450633] RIP: 0033:0x786d7540c88d
[    1.450638] Code: ff c3 66 2e 0f 1f 84 00 00 00 00 00 90 f3 0f 1e fa 48 89 f8 48 89 f7 48 89 d6 48 89 ca 4d 89 c2 4d 89 c8 4c 8b 4c 24 08 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 53 04 0d 00 f7 d8 64 89 01 48
[    1.450641] RSP: 002b:00007ffdfa693b18 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
[    1.450645] RAX: ffffffffffffffda RBX: 00005b8d3bb39e70 RCX: 0000786d7540c88d
[    1.450647] RDX: 0000000000000004 RSI: 0000786d74c042f2 RDI: 000000000000001e
[    1.450649] RBP: 00007ffdfa693bb0 R08: 0000000000000000 R09: 00005b8d3bb3ef00
[    1.450651] R10: 0000000000000000 R11: 0000000000000246 R12: 0000786d74c042f2
[    1.450653] R13: 0000000000020000 R14: 00005b8d3bb3ad30 R15: 00005b8d3bb39e70
[    1.450657]  </TASK>
[    1.450659] ---[ end trace 0000000000000000 ]---
```

lgirdwood, 2025-03-12:
@ujfalusi is there a kernel option that fixes this audio warning about HDMI converter 3?
```
[    9.029398] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 2:2:0-57864
[    9.029405] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:22:1 Kernel ABI 3:23:1
[    9.041412] sof-audio-pci-intel-tgl 0000:00:1f.3: Topology: ABI 3:22:1 Kernel ABI 3:23:1
[    9.041717] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: Parent card not yet available, widget card binding deferred
[    9.068118] snd_hda_codec_realtek ehdaudio0D0: ALC257: picked fixup  for PCI SSID 17aa:0000
[    9.068470] snd_hda_codec_realtek ehdaudio0D0: autoconfig for ALC257: line_outs=1 (0x14/0x0/0x0/0x0/0x0) type:speaker
[    9.068474] snd_hda_codec_realtek ehdaudio0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[    9.068476] snd_hda_codec_realtek ehdaudio0D0:    hp_outs=1 (0x21/0x0/0x0/0x0/0x0)
[    9.068477] snd_hda_codec_realtek ehdaudio0D0:    mono: mono_out=0x0
[    9.068479] snd_hda_codec_realtek ehdaudio0D0:    inputs:
[    9.068480] snd_hda_codec_realtek ehdaudio0D0:      Mic=0x19
[    9.106784] skl_hda_dsp_generic skl_hda_dsp_generic: hda_dsp_hdmi_build_controls: no PCM in topology for HDMI converter 3
```

khpylon, 2025-03-12:
This is the systemd service I use:
```
[Unit]
Description=Disable and reconnect Intel Raptor Lake-P/U/H cAVS sound device on hibernation
Before=sleep.target
StopWhenUnneeded=yes

[Service]
User=root
Type=oneshot
RemainAfterExit=yes
ExecStart=/usr/local/bin/remove-soundcard.bash
ExecStop=/usr/bin/sleep 0.5s
ExecStop=/usr/bin/bash -c "echo 1 > /sys/bus/pci/rescan"

[Install]
WantedBy=sleep.target
```

and /usr/local/bin/remove-soundcard.bash:
```
#!/bin/bash
DEVICE=$(lspci -D | grep 'Raptor Lake-P/U/H cAVS' | cut -d' ' -f1)
if [[ $DEVICE != "" ]] ; then 
  echo 1 > /sys/bus/pci/devices/$DEVICE/remove
  sleep 1
fi
```

lgirdwood, 2025-03-14:
@khpylon thanks for the script, I do think there is an interaction between audio and display driver at suspend and it likely to be related to the warning being reported in the display driver. Btw, did this work with a previous kernel and has the GFW warning always been display no matter which kernel was used ?

Btw, we do test audio and display within our CI with this version on IP and have not seen this issue within our configuration.

One thing to try, can you make sure all system FW/BIOS is up to date by running ```fwupdmgr``` as this can sometimes help with issues the kernel is reporting about the BIOS.

khpylon, 2025-03-15:
This is a laptop I just bought in November, and as far as I know it's always done this.   I checked the BIOS and ran fwupdmgr.  Both are up to date.  

The line in intel_bios.c that is crashing is the drm_WARN call below, and the comment in the code says a lot:

```
	/*
	 * FIXME need to implement support for VBT
	 * vswing/preemph tables should this ever trigger.
	 */
	drm_WARN(display->drm, child->use_vbt_vswing,
		 "Port %c asks to use VBT vswing/preemph tables\n",
		 port_name(port));
```

Google found [this reference](https://lore.kernel.org/all/YxB6Xa+KUwoWCZf8@intel.com/)

I removed that call and built the kernel module.  The problem still exists, so it's not the actual crash causing it.

kv2019i, 2025-03-18:
Moving to kernel, this is not a SOF FW bug.