thesofproject/sof #9908: [BUG] HP Omnibook Ultra Flip: sound from speakers when headphones plugged in

qwence, 2025-03-18:
**Describe the bug**
When I plug my headphones (connected via normal headphone jack) into my HP Omnibook Ultra Flip 14 laptop (which uses a Lunar Lake chip), sound comes out of the speakers as well as the headphones. I'm using pipewire-pulse and alsa, and alsamixer shows that the speakers are muted when the headphones are plugged in. `pavucontrol` shows that the output switches from speakers to headphones. This works as expected on windows, and also the headphones work fine on my other linux laptop.

**Reproduction Rate**
It happens every time.

**Expected behavior**
When headphones are plugged in, sound comes out of the headphones but not the speakers.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 6.13.7-arch1-1
    * SOF: 2.12
2) Name of the topology file
    * Not sure
3) Name of the platform(s) on which the bug is observed.
    * Platform: Arch Linux

**Screenshots or console output**
Relevant parts of dmesg:
```
[   11.015543] snd_hda_intel 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[   11.172637] sof-audio-pci-intel-lnl 0000:00:1f.3: enabling device (0000 -> 0002)
[   11.172731] sof-audio-pci-intel-lnl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[   11.172920] sof-audio-pci-intel-lnl 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [xe])
[   11.179649] sof-audio-pci-intel-lnl 0000:00:1f.3: use msi interrupt mode
[   11.203420] sof-audio-pci-intel-lnl 0000:00:1f.3: hda codecs found, mask 5
[   11.203427] sof-audio-pci-intel-lnl 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[   11.203429] sof-audio-pci-intel-lnl 0000:00:1f.3: NHLT device BT(0) detected, ssp_mask 0x4
[   11.203430] sof-audio-pci-intel-lnl 0000:00:1f.3: BT link detected in NHLT tables: 0x4
[   11.203432] sof-audio-pci-intel-lnl 0000:00:1f.3: DMICs detected in NHLT tables: 2
[   11.208173] sof-audio-pci-intel-lnl 0000:00:1f.3: Firmware paths/files for ipc type 1:
[   11.208178] sof-audio-pci-intel-lnl 0000:00:1f.3:  Firmware file:     intel/sof-ipc4/lnl/sof-lnl.ri
[   11.208179] sof-audio-pci-intel-lnl 0000:00:1f.3:  Firmware lib path: intel/sof-ipc4-lib/lnl
[   11.208180] sof-audio-pci-intel-lnl 0000:00:1f.3:  Topology file:     intel/sof-ipc4-tplg/sof-hda-generic-2ch.tplg
[   11.208678] sof-audio-pci-intel-lnl 0000:00:1f.3: Loaded firmware library: ADSPFW, version: 2.12.0.1
[   11.496232] sof-audio-pci-intel-lnl 0000:00:1f.3: Booted firmware version: 2.12.0.1
[   11.521560] sof-audio-pci-intel-lnl 0000:00:1f.3: Topology: ABI 3:29:1 Kernel ABI 3:23:1
[   11.529439] sof-audio-pci-intel-lnl 0000:00:1f.3: Loaded firmware library: ADSPFW, version: 2.12.0.1
[   11.631069] snd_hda_codec_realtek ehdaudio0D0: ALC245: picked fixup  for PCI SSID 103c:8cde
[   12.052460] snd_hda_codec_realtek ehdaudio0D0: bound i2c-CSC3551:00-cs35l41-hda.0 (ops cs35l41_hda_comp_ops [snd_hda_scodec_cs35l41])
[   12.463551] snd_hda_codec_realtek ehdaudio0D0: bound i2c-CSC3551:00-cs35l41-hda.1 (ops cs35l41_hda_comp_ops [snd_hda_scodec_cs35l41])
[   12.463736] snd_hda_codec_realtek ehdaudio0D0: autoconfig for ALC245: line_outs=1 (0x17/0x0/0x0/0x0/0x0) type:speaker
[   12.463738] snd_hda_codec_realtek ehdaudio0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[   12.463740] snd_hda_codec_realtek ehdaudio0D0:    hp_outs=1 (0x21/0x0/0x0/0x0/0x0)
[   12.463740] snd_hda_codec_realtek ehdaudio0D0:    mono: mono_out=0x0
[   12.463741] snd_hda_codec_realtek ehdaudio0D0:    inputs:
[   12.463742] snd_hda_codec_realtek ehdaudio0D0:      Mic=0x19
[   12.532934] input: sof-hda-dsp Mic as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input34
[   12.532970] input: sof-hda-dsp Headphone as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input35
[   12.532996] input: sof-hda-dsp HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input36
[   12.533021] input: sof-hda-dsp HDMI/DP,pcm=4 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input37
[   12.533052] input: sof-hda-dsp HDMI/DP,pcm=5 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input38
[   12.813601] usbcore: registered new interface driver snd-usb-audio
```

alsa-info.sh (run when the headphones were plugged in)
```
upload=true&script=true&cardinfo=
!!################################
!!ALSA Information Script v 0.5.3
!!################################

!!Script ran on: Tue Mar 18 23:22:41 UTC 2025

!!Linux Distribution
!!------------------

NAME="Arch Linux" PRETTY_NAME="Arch Linux" ID=arch HOME_URL="https://archlinux.org/" DOCUMENTATION_URL="https://wiki.archlinux.org/" SUPPORT_URL="https://bbs.archlinux.org/" BUG_REPORT_URL="https://gitlab.archlinux.org/groups/archlinux/-/issues" PRIVACY_POLICY_URL="https://terms.archlinux.org/docs/privacy-policy/" LOGO=archlinux-logo

[... 1820 lines cut ...]
0x04 0x18560010
0x06 0x18560010
0x08 0x18560010
0x0a 0x18560010
0x0b 0x18560010
0x0c 0x18560010
0x0d 0x18560010
0x0e 0x18560010
0x0f 0x18560010

/sys/class/sound/hwC0D2/driver_pin_configs:

/sys/class/sound/hwC0D2/user_pin_configs:

/sys/class/sound/hwC0D2/init_verbs:

/sys/class/sound/hwC0D2/hints:

!!ALSA/HDA dmesg
!!--------------

!!Packages installed
!!--------------------

```

lgirdwood, 2025-03-19:
Adding @ujfalusi who has recently solved a similar issue with HDA  spk/hp switch. @qwence this is a userspace issue, where the UCM configuration is not muting the speaker via alsamixer (you should be able to manually mute in alsamixer for the time being)

qwence, 2025-03-20:
When I plug in headphones, alsamixer already reports the speaker as muted.

@ujfalusi Let me know if you have any ideas, or if there is a more appropriate place for me to report this (I wasn't sure).

ujfalusi, 2025-03-20:
@qwence, the first thing we ask is to confirm the behavior with the legacy HDA stack by adding
`options snd_intel_dspcfg dsp_driver=1` to for example `/etc/modprobe.d/force-legacy-hda.conf` and reboot
You will loose DMIC, but you should be able to verify the Speaker/Headphone issue.

Have you tried to manually change the mixer controls (`alsamixer -c0`) while playing audio and having the headset connected?
Is there a combination when the audio is routed correctly?

It looks like the device needs some quirk to get it's configuration right.

Can you also share the output of `amixer -c0 contents` and `amixer -c0 scontents` ?

bardliao, 2025-03-20:
@KailangYang Could you help on it?

ujfalusi, 2025-03-20:
One thing that is interesting in alsa-info is that it presents two Jack function ('Mic Jack' and 'Headphone Jack') while the laptop has a single combination jack for headphone/headset/mic, likely unrelated to the speaker is always on issue, but hints of mis-identification of the setup.

qwence, 2025-03-20:
`amixer -c0 contents`

[amixer_contents.txt](https://github.com/user-attachments/files/19368226/amixer_contents.txt)

`amixer -c0 scontents`

[amixer_scontents.txt](https://github.com/user-attachments/files/19368230/amixer_scontents.txt)

I tried the legacy HDA stack as suggested and it resulted in the same behavior.

When I do `alsamixer -c0` with the headphones plugged in, it shows the speaker as muted. I can use the gui to unmute the speaker, change the volume to zero, and remute it, and none of this changes the sound coming out of the speakers. I also tried changing the other settings and they don't affect the problem; various settings will raise/lower the volume, and they affect both the headphone sound and the unwanted speaker sound.

One other observation is that the speaker sound does change when the headphones are plugged in: it is less loud, but also different in quality-- less full or less bass? (Sorry, I'm the exact opposite of an audiophile.) I wonder if there are multiple speakers and only one is muted. (`alsamixer` does only show one entry for "speaker".)

ujfalusi, 2025-03-20:
Looks like Quad speaker: https://www.hp.com/us-en/shop/pdp/HP-OmniBook-Ultra-Flip-14---fh0047nr#techSpecs