thesofproject/sof #9914: [BUG] "IPC timed out" on multiple ACE platforms with xrun warnings in mtrace

lyakh, 2025-03-21:
**Describe the bug**
multiple-pause-resume-50 and multiple-pipeline-playback tests fail on PTL and LNL with xrun floods in mtrace logs, leading to IPC timeouts.

Example 1:
https://sof-ci.01.org/sofpr/PR9893/build11697/devicetest/index.html?model=PTLH_HDA_AIOC&testcase=multiple-pipeline-playback
platform: PTL
test: multiple-pipeline-playback
mtrace:
[00:00:05.699,326] <wrn> dai_comp: dai_common_copy: comp:1 0x10003 nothing to copy, src_frames: 96, sink_frames: 0
Linux kernel:
[ 1129.969814] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-ptl 0000:00:1f.3: ipc tx      : 0x13000003|0x0: GLB_SET_PIPELINE_STATE
[ 1130.473670] kernel: sof-audio-pci-intel-ptl 0000:00:1f.3: ipc timed out for 0x13000003|0x0

Example 2:
https://sof-ci.ostc.intel.com/#/result/planresultdetail/50247?model=LNLM_RVP_HDA&testcase=multiple-pause-resume-50
platform: LNL
test: multiple-pause-resume-50
mtrace:
[  763.205285] <inf> host_comp: host_get_copy_bytes_normal: comp:3 0x40004 no bytes to copy, available samples: 0, free_samples: 768
Linux kernel:
[  762.318154] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-lnl 0000:00:1f.3: ipc tx      : 0x12010000|0x0: GLB_DELETE_PIPELINE
[  762.824476] kernel: sof-audio-pci-intel-lnl 0000:00:1f.3: ipc timed out for 0x12010000|0x0

Example 3 (uncertain):
https://sof-ci.ostc.intel.com/#/result/planresultdetail/50754?model=LNLM_SDW_AIOC&testcase=multiple-pause-resume-50
platform: LNL
test: multiple-pause-resume-50
mtrace:
[ 1439.250870] <wrn> dai_comp: dai_common_copy: comp:0 0x4 nothing to copy, src_frames: 0, sink_frames: 96
Linux kernel:
---
Uncertain because the firmware flooding error is different, configuration is different (SDW vs. HDA), no kernel logs to check IPC timeouts

**To Reproduce**
So far observed with multiple-pause-resume-50 and multiple-pipeline-playback tests on LNL and PTL platforms in HDA configurations

**Reproduction Rate**
Rare

**Impact**
Must be fixed

**Environment**
The oldest of the above examples is from Feb 26 2025

lgirdwood, 2025-03-24:
@lyakh can you confirm if the host no bytes errors are for playback or capture ? If playback have we seen this with capture ?

lyakh, 2025-03-24:
another instance of this bug https://sof-ci.01.org/sofpr/PR9921/build11752/devicetest/index.html?model=PTLP_RVP_SDW&testcase=multiple-pipeline-all

lyakh, 2025-03-24:
> [@lyakh](https://github.com/lyakh) can you confirm if the host no bytes errors are for playback or capture ? If playback have we seen this with capture ?

@lgirdwood here's a good capture - with synchronised timestamps, both kernel and firmware logs available:
```
[ 2339.493920] kernel: snd_sof:sof_widget_free_unlocked: sof-audio-pci-intel-ptl 0000:00:1f.3: widget dai-copier.DMIC.dmic01.capture freed
[ 2339.493938] kernel: snd_sof:sof_ipc4_route_free: sof-audio-pci-intel-ptl 0000:00:1f.3: unbind modules gain.101.1:0 -> eqiir.101.1:0
[ 2339.493947] kernel: snd_sof:sof_widget_free_unlocked: sof-audio-pci-intel-ptl 0000:00:1f.3: widget gain.101.1 freed
[ 2339.493951] kernel: snd_sof:sof_ipc4_route_free: sof-audio-pci-intel-ptl 0000:00:1f.3: unbind modules eqiir.101.1:0 -> module-copier.101.2:0
[ 2339.493954] kernel: snd_sof:sof_widget_free_unlocked: sof-audio-pci-intel-ptl 0000:00:1f.3: widget eqiir.101.1 freed
[ 2339.493958] kernel: snd_sof:sof_ipc4_route_free: sof-audio-pci-intel-ptl 0000:00:1f.3: unbind modules module-copier.101.2:0 -> host-copier.10.capture:0
[ 2339.493964] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-ptl 0000:00:1f.3: ipc tx      : 0x460b0002|0xc0002: MOD_UNBIND
[ 2339.494311] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-ptl 0000:00:1f.3: ipc tx reply: 0x66000000|0xc0002: MOD_UNBIND
[ 2339.494381] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-ptl 0000:00:1f.3: ipc tx done : 0x460b0002|0xc0002: MOD_UNBIND
[ 2339.494405] kernel: snd_sof:sof_ipc4_log_header: sof-audio-pci-intel-ptl 0000:00:1f.3: ipc tx      : 0x12090000|0x0: GLB_DELETE_PIPELINE
[ 2339.494635] kernel: snd_sof:sof_pcm_trigger: sof-audio-pci-intel-ptl 0000:00:1f.3: pcm1 (Jack In), dir 1: Entry: trigger (cmd: 0)
[ 2339.494642] kernel: snd_sof:sof_ipc4_trigger_pipelines: sof-audio-pci-intel-ptl 0000:00:1f.3: pcm1 (Jack In), dir 1: cmd: 0, state: 3
[ 2339.515680] kernel: snd_sof:sof_pcm_trigger: sof-audio-pci-intel-ptl 0000:00:1f.3: pcm6 (HDMI2), dir 0: Entry: trigger (cmd: 0)
[ 2339.515686] kernel: snd_sof:sof_ipc4_trigger_pipelines: sof-audio-pci-intel-ptl 0000:00:1f.3: pcm6 (HDMI2), dir 0: cmd: 0, state: 3
[ 2339.515693] kernel: snd_sof:sof_pcm_trigger: sof-audio-pci-intel-ptl 0000:00:1f.3: pcm5 (HDMI1), dir 0: Entry: trigger (cmd: 0)
[ 2339.515701] kernel: snd_sof:sof_ipc4_trigger_pipelines: sof-audio-pci-intel-ptl 0000:00:1f.3: pcm5 (HDMI1), dir 0: cmd: 0, state: 3
[ 2339.515749] kernel: snd_sof:sof_pcm_trigger: sof-audio-pci-intel-ptl 0000:00:1f.3: pcm2 (Speaker), dir 0: Entry: trigger (cmd: 0)
[ 2339.515755] kernel: snd_sof:sof_ipc4_trigger_pipelines: sof-audio-pci-intel-ptl 0000:00:1f.3: pcm2 (Speaker), dir 0: cmd: 0, state: 3
[ 2339.515764] kernel: snd_sof:sof_pcm_trigger: sof-audio-pci-intel-ptl 0000:00:1f.3: pcm0 (Jack Out), dir 0: Entry: trigger (cmd: 0)
[ 2339.515770] kernel: snd_sof:sof_pcm_trigger: sof-audio-pci-intel-ptl 0000:00:1f.3: pcm7 (HDMI3), dir 0: Entry: trigger (cmd: 0)
[ 2339.515770] kernel: snd_sof:sof_ipc4_trigger_pipelines: sof-audio-pci-intel-ptl 0000:00:1f.3: pcm0 (Jack Out), dir 0: cmd: 0, state: 3
[ 2339.515776] kernel: snd_sof:sof_ipc4_trigger_pipelines: sof-audio-pci-intel-ptl 0000:00:1f.3: pcm7 (HDMI3), dir 0: cmd: 0, state: 3
[ 2339.515834] kernel: snd_sof:sof_pcm_trigger: sof-audio-pci-intel-ptl 0000:00:1f.3: pcm31 (Deepbuffer Jack Out), dir 0: Entry: trigger (cmd: 0)
[ 2339.515840] kernel: snd_sof:sof_ipc4_trigger_pipelines: sof-audio-pci-intel-ptl 0000:00:1f.3: pcm31 (Deepbuffer Jack Out), dir 0: cmd: 0, state: 3
[ 2340.002557] kernel: sof-audio-pci-intel-ptl 0000:00:1f.3: ipc timed out for 0x12090000|0x0
[ 2340.002607] kernel: sof-audio-pci-intel-ptl 0000:00:1f.3: Attempting to prevent DSP from entering D3 state to preserve context
[ 2340.002626] kernel: sof-audio-pci-intel-ptl 0000:00:1f.3: ------------[ IPC dump start ]------------
[ 2340.002654] kernel: sof-audio-pci-intel-ptl 0000:00:1f.3: Host IPC initiator: 0x92090000|0x0|0x0, target: 0x1b060000|0x0|0x0, ctl: 0x3
[ 2340.002677] kernel: sof-audio-pci-intel-ptl 0000:00:1f.3: ------------[ IPC dump end ]------------
```
That seems to be capture, but I'm not 100% certain that DMIC messages above the timeout error are related. The corresponding firmware part:
```
[ 2339.494415] <inf> ipc: ipc_cmd: rx	: 0x12090000|0x0
[ 2339.494456] <inf> dma: sof_dma_put: dma_put(), dma = 0x4009e2e0, sref = 2
[ 2339.494463] <inf> dai_intel_dmic: dai_dmic_remove: dmic_remove(), dmic_active_fifos_mask = 0x0, dmic_pause_mask = 0x0
[ 2339.495076] <inf> host_comp: host_get_copy_bytes_normal: comp:5 0x60002 no bytes to copy, available samples: 0, free_samples: 384
[ 2339.496075] <inf> host_comp: host_get_copy_bytes_normal: comp:5 0x60002 no bytes to copy, available samples: 0, free_samples: 384
[ 2339.497075] <inf> host_comp: host_get_copy_bytes_normal: comp:5 0x60002 no bytes to copy, available samples: 0, free_samples: 384
```
The test was trying to run 6 playback and 3 capture streams simultaneously So, hard to be sure...

lyakh, 2025-03-25:
Currently it seems that this bug is almost all PTL SDW jenkins test runs fail the multiple-pipeline-all.sh test, however, some do pass, e.g. https://sof-ci.01.org/linuxpr/PR5366/build6770/devicetest/index.html?model=PTLP_RVP_SDW&testcase=multiple-pipeline-all

lyakh, 2025-04-22:
no reproduction found since #9926 and #9929 got merged, propose to close