thesofproject/sof #9920: [BUG]

pankhurijain20, 2025-03-23:
no sound coming from Lenovo on Ubuntu 24.04

    Your laptop model: Lenovo IdeaPad Pro 5 14IMH9

    Output of:

    dmesg | grep sof
    aplay -l
    uname -a

    You're on Ubuntu 24.04 with kernel 6.11+

    SOF loads correctly, but no sound from speakers or headphones

    Even aplay with plughw:0,0 gives Device or resource busy or silence

lgirdwood, 2025-03-24:
@pankhurijain20 can you provide the logs. They are missing, you have just provided the commands only.

pankhurijain20, 2025-03-24:
Please find the logs

dmesg: read kernel buffer failed: Operation not permitted
**** List of PLAYBACK Hardware Devices ****
card 0: sofhdadsp [sof-hda-dsp], device 0: HDA Analog (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 3: HDMI1 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 4: HDMI2 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 5: HDMI3 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 31: Deepbuffer HDA Analog (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
Linux pankhuri-IdeaPad-Pro-5-14IMH9 6.11.0-17-generic #17~24.04.2-Ubuntu SMP PREEMPT_DYNAMIC Mon Jan 20 22:48:29 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
pankhuri@pankhuri-IdeaPad-Pro-5-14IMH9:~/workspace

lgirdwood, 2025-03-24:
@pankhurijain20 still missing kernel logs.

ujfalusi, 2025-03-25:
@pankhurijain20, can you provide the output of `alsa-info.sh` as first step, then can you try to switch to the legacy HDA stack by adding `options snd_intel_dspcfg dsp_driver=1` to for example `/etc/modprobe.d/force-legacy-hda.conf` and reboot
You will loose DMIC, but you should be able to verify the Speaker/Headphone issue.

pankhurijain20, 2025-03-25:
ALSA Information Script v 0.5.3
--------------------------------

This script visits the following commands/files to collect diagnostic
information about your ALSA installation and sound related hardware.

  dmesg
  lspci
  aplay
  amixer
  alsactl
  rpm, dpkg
  /proc/asound/
  /sys/class/sound/
  ~/.asoundrc (etc.)

See 'bash --help' for command line options.

cat: '/sys/module/snd_soc_skl_hda_dsp/parameters/*': No such file or directory
dmesg: read kernel buffer failed: Operation not permitted
Automatically upload ALSA information to www.alsa-project.org? [y/N] : y
Uploading information to www.alsa-project.org ... Done!

Your ALSA information is located at http://alsa-project.org/db/?f=3704c536e96ba4d32d03b34e18ee1174355a1e55
Please inform the person helping you.

pankhurijain20, 2025-03-25:
Actually I dont have /etc/modprobe.d/force-legacy-hda.conf

ujfalusi, 2025-03-25:
> Actually I dont have /etc/modprobe.d/force-legacy-hda.conf

yes, you need to create it, then when tested the speaker/headset remove it.

pankhurijain20, 2025-03-25:
Still its not working