thesofproject/sof #9930: [BUG] Missing topology file `sof-lnl-cs42l43-l0-cs35l56-l23-4ch.tplg` on Zenbook S 16 with kernel 6.13.8

PisonJay, 2025-03-27:
**Describe the bug**
On Zenbook S 16 with kernel 6.13.8, the dmesg reports missing topology file `sof-lnl-cs42l43-l0-cs35l56-l23-4ch.tplg`

**To Reproduce**
Boot the system and run `sudo dmesg | grep audio` to show logs below
```
❯ sudo dmesg | grep audio
[   11.643538] sof-audio-pci-intel-lnl 0000:00:1f.3: enabling device (0000 -> 0002)
[   11.643893] sof-audio-pci-intel-lnl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[   11.643979] sof-audio-pci-intel-lnl 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [xe])
[   11.650495] sof-audio-pci-intel-lnl 0000:00:1f.3: use msi interrupt mode
[   11.669211] sof-audio-pci-intel-lnl 0000:00:1f.3: hda codecs found, mask 4
[   11.669219] sof-audio-pci-intel-lnl 0000:00:1f.3: NHLT device BT(0) detected, ssp_mask 0x4
[   11.669222] sof-audio-pci-intel-lnl 0000:00:1f.3: BT link detected in NHLT tables: 0x4
[   11.669226] sof-audio-pci-intel-lnl 0000:00:1f.3: DMICs detected in NHLT tables: 4
[   11.674425] sof-audio-pci-intel-lnl 0000:00:1f.3: SOF firmware and/or topology file not found.
[   11.674436] sof-audio-pci-intel-lnl 0000:00:1f.3: Supported default profiles
[   11.674438] sof-audio-pci-intel-lnl 0000:00:1f.3: - ipc type 1 (Requested):
[   11.674442] sof-audio-pci-intel-lnl 0000:00:1f.3:  Firmware file: intel/sof-ipc4/lnl/sof-lnl.ri
[   11.674444] sof-audio-pci-intel-lnl 0000:00:1f.3:  Topology file: intel/sof-ipc4-tplg/sof-lnl-cs42l43-l0-cs35l56-l23-4ch.tplg
[   11.674446] sof-audio-pci-intel-lnl 0000:00:1f.3: Check if you have 'sof-firmware' package installed.
[   11.674448] sof-audio-pci-intel-lnl 0000:00:1f.3: Optionally it can be manually downloaded from:
[   11.674449] sof-audio-pci-intel-lnl 0000:00:1f.3:    https://github.com/thesofproject/sof-bin/
[   18.227549] sof-audio-pci-intel-lnl 0000:00:1f.3: error: sof_probe_work failed err: -2
```

**Reproduction Rate**
All the time

**Expected behavior**
The topology should exist and work.

**Impact**
No audio device can be found

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: stable 6.13.8
    * SOF: sof-bin release v2025.01
2) Name of the topology file
    * Topology: intel/sof-ipc4-tplg/sof-lnl-cs42l43-l0-cs35l56-l23-4ch.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: ASUS Zenbook S 16 UX5606SA with LunarLake Ultra 7 258V

**Screenshots or console output**

[image]

PisonJay, 2025-03-27:
There are related issues at https://github.com/thesofproject/linux/issues/5325 and https://github.com/thesofproject/sof-bin/issues/177 .
I tried building new tplg with patch below

```patch
diff --git a/tools/topology/topology2/production/tplg-targets-ace2.cmake b/tools/topology/topology2/production/tplg-targets-ace2.cmake
index 12a75b3bb..95969507c 100644
--- a/tools/topology/topology2/production/tplg-targets-ace2.cmake
+++ b/tools/topology/topology2/production/tplg-targets-ace2.cmake
@@ -91,6 +91,13 @@ DMIC0_ID=3,DMIC1_ID=4,HDMI1_ID=5,HDMI2_ID=6,HDMI3_ID=7,DMIC0_ENHANCED_CAPTURE=tr
 EFX_DMIC0_TDFB_PARAMS=line2_generic_pm10deg,EFX_DMIC0_DRC_PARAMS=dmic_default,\
 PREPROCESS_PLUGINS=nhlt,NHLT_BIN=sof-lnl-cs42l43-l0-cs35l56-l23-2ch.bin"
 
+"cavs-sdw\;sof-lnl-cs42l43-l0-cs35l56-l23-4ch\;PLATFORM=lnl,NUM_SDW_AMP_LINKS=2,\
+SDW_AMP_FEEDBACK=false,SDW_SPK_STREAM=Playback-SmartAmp,\
+SDW_JACK_OUT_STREAM=Playback-SimpleJack,SDW_JACK_IN_STREAM=Capture-SimpleJack,NUM_DMICS=4,\
+PDM1_MIC_A_ENABLE=1,PDM1_MIC_B_ENABLE=1,DMIC0_ID=3,DMIC1_ID=4,HDMI1_ID=5,HDMI2_ID=6,HDMI3_ID=7,DMIC0_ENHANCED_CAPTURE=true,\
+EFX_DMIC0_TDFB_PARAMS=line4_pass,EFX_DMIC0_DRC_PARAMS=dmic_default,\
+PREPROCESS_PLUGINS=nhlt,NHLT_BIN=sof-lnl-cs42l43-l0-cs35l56-l23-4ch.bin"
+
 "cavs-sdw\;sof-lnl-cs42l43-l2-cs35l56x6-l13\;NUM_SDW_AMP_LINKS=3,SDW_DMIC=1,\
 SDW_AMP_FEEDBACK=false,SDW_SPK_STREAM=Playback-SmartAmp,SDW_DMIC_STREAM=Capture-SmartMic,\
 SDW_JACK_OUT_STREAM=Playback-SimpleJack,SDW_JACK_IN_STREAM=Capture-SimpleJack"
```

But it brings new error that physical link dmic01 (id 3) not exist.

```
❯ sudo dmesg | grep -E '(audio|sof)'
[    0.043901] software IO TLB: area num 8.
[    0.725487] PCI-DMA: Using software bounce buffering for IO (SWIOTLB)
[    0.725488] software IO TLB: mapped [mem 0x0000000050001000-0x0000000054001000] (64MB)
[   10.054389] sof-audio-pci-intel-lnl 0000:00:1f.3: enabling device (0000 -> 0002)
[   10.054524] sof-audio-pci-intel-lnl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[   10.054565] sof-audio-pci-intel-lnl 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [xe])
[   10.061475] sof-audio-pci-intel-lnl 0000:00:1f.3: use msi interrupt mode
[   10.079641] sof-audio-pci-intel-lnl 0000:00:1f.3: hda codecs found, mask 4
[   10.079645] sof-audio-pci-intel-lnl 0000:00:1f.3: NHLT device BT(0) detected, ssp_mask 0x4
[   10.079646] sof-audio-pci-intel-lnl 0000:00:1f.3: BT link detected in NHLT tables: 0x4
[   10.079648] sof-audio-pci-intel-lnl 0000:00:1f.3: DMICs detected in NHLT tables: 4
[   10.082147] sof-audio-pci-intel-lnl 0000:00:1f.3: Firmware paths/files for ipc type 1:
[   10.082149] sof-audio-pci-intel-lnl 0000:00:1f.3:  Firmware file:     intel/sof-ipc4/lnl/sof-lnl.ri
[   10.082149] sof-audio-pci-intel-lnl 0000:00:1f.3:  Firmware lib path: intel/sof-ipc4-lib/lnl
[   10.082150] sof-audio-pci-intel-lnl 0000:00:1f.3:  Topology file:     intel/sof-ipc4-tplg/sof-lnl-cs42l43-l0-cs35l56-l23-4ch.tplg
[   10.082446] sof-audio-pci-intel-lnl 0000:00:1f.3: Loaded firmware library: ADSPFW, version: 2.12.0.1
[   10.344345] sof-audio-pci-intel-lnl 0000:00:1f.3: Booted firmware version: 2.12.0.1
[   10.354986] snd_soc_sdw_utils:asoc_sdw_count_sdw_endpoints: sof_sdw sof_sdw: Found 5 devices with 8 endpoints
[   10.354991] snd_soc_sdw_utils:asoc_sdw_parse_sdw_endpoints: sof_sdw sof_sdw: Adding prefix cs42l43 for cs42l43-codec
[   10.354993] snd_soc_sdw_utils:asoc_sdw_parse_sdw_endpoints: sof_sdw sof_sdw: Add dev: 0, 0x3001fa424301 end: 0, dai: 0, P/- to solo: 0
[   10.354995] snd_soc_sdw_utils:asoc_sdw_parse_sdw_endpoints: sof_sdw sof_sdw: Add dev: 0, 0x3001fa424301 end: 1, dai: 2, -/C to solo: 0
[   10.354997] snd_soc_sdw_utils:asoc_sdw_parse_sdw_endpoints: sof_sdw sof_sdw: Add dev: 0, 0x3001fa424301 end: 2, dai: 0, -/C to solo: 0
[   10.354999] snd_soc_sdw_utils:asoc_sdw_parse_sdw_endpoints: sof_sdw sof_sdw: Adding prefix AMP1 for sdw:0:2:01fa:3556:01:0
[   10.355000] snd_soc_sdw_utils:asoc_sdw_parse_sdw_endpoints: sof_sdw sof_sdw: Add dev: 2, 0x23001fa355601 end: 0, dai: 1, P/C to group: 1
[   10.355002] snd_soc_sdw_utils:asoc_sdw_parse_sdw_endpoints: sof_sdw sof_sdw: Adding prefix AMP2 for sdw:0:2:01fa:3556:01:1
[   10.355003] snd_soc_sdw_utils:asoc_sdw_parse_sdw_endpoints: sof_sdw sof_sdw: Add dev: 2, 0x23101fa355601 end: 0, dai: 1, P/C to group: 1
[   10.355004] snd_soc_sdw_utils:asoc_sdw_parse_sdw_endpoints: sof_sdw sof_sdw: Adding prefix AMP3 for sdw:0:3:01fa:3556:01:2
[   10.355005] snd_soc_sdw_utils:asoc_sdw_parse_sdw_endpoints: sof_sdw sof_sdw: Add dev: 3, 0x33201fa355601 end: 0, dai: 1, P/C to group: 1
[   10.355006] snd_soc_sdw_utils:asoc_sdw_parse_sdw_endpoints: sof_sdw sof_sdw: Adding prefix AMP4 for sdw:0:3:01fa:3556:01:3
[   10.355007] snd_soc_sdw_utils:asoc_sdw_parse_sdw_endpoints: sof_sdw sof_sdw: Add dev: 3, 0x33301fa355601 end: 0, dai: 1, P/C to group: 1
[   10.355009] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link SDW0-Playback-SimpleJack, id 0
[   10.355011] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link SDW0-Capture-SmartMic, id 4
[   10.355013] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link SDW0-Capture-SimpleJack, id 1
[   10.355014] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link SDW2-Playback-SmartAmp, id 2
[   10.355016] snd_soc_sdw_utils:asoc_sdw_init_dai_link: sof_sdw sof_sdw: create dai link SDW2-Capture-SmartAmp, id 3
[... cut ...]
                        "direction-angle":90
                },
                {
                        "channel":3,
                        "type":"omnidirectional",
                        "location":"laptop-left-panel",
                        "speaker-distance":48,
                        "horizontal-offset":-195,
                        "vertical-offset":-230,
                        "freq-low-band":100,
                        "freq-high-band":20000,
                        "direction-angle":-90
                }
        ]
}
```

ujfalusi, 2025-03-27:
@PisonJay, can you try to build the tplg with `DMIC0_ID=5,DMIC1_ID=6` ?

PisonJay, 2025-03-27:
```
diff --git a/tools/topology/topology2/production/tplg-targets-ace2.cmake b/tools/topology/topology2/production/tplg-targets-ace2.cmake
index 12a75b3bb..43559efe6 100644
--- a/tools/topology/topology2/production/tplg-targets-ace2.cmake
+++ b/tools/topology/topology2/production/tplg-targets-ace2.cmake
@@ -91,6 +91,13 @@ DMIC0_ID=3,DMIC1_ID=4,HDMI1_ID=5,HDMI2_ID=6,HDMI3_ID=7,DMIC0_ENHANCED_CAPTURE=tr
 EFX_DMIC0_TDFB_PARAMS=line2_generic_pm10deg,EFX_DMIC0_DRC_PARAMS=dmic_default,\
 PREPROCESS_PLUGINS=nhlt,NHLT_BIN=sof-lnl-cs42l43-l0-cs35l56-l23-2ch.bin"
 
+"cavs-sdw\;sof-lnl-cs42l43-l0-cs35l56-l23-4ch\;PLATFORM=lnl,NUM_SDW_AMP_LINKS=2,\
+SDW_AMP_FEEDBACK=false,SDW_SPK_STREAM=Playback-SmartAmp,\
+SDW_JACK_OUT_STREAM=Playback-SimpleJack,SDW_JACK_IN_STREAM=Capture-SimpleJack,NUM_DMICS=4,\
+PDM1_MIC_A_ENABLE=1,PDM1_MIC_B_ENABLE=1,DMIC0_ID=5,DMIC1_ID=6,HDMI1_ID=7,HDMI2_ID=8,HDMI3_ID=9,DMIC0_ENHANCED_CAPTURE=true,\
+EFX_DMIC0_TDFB_PARAMS=line4_pass,EFX_DMIC0_DRC_PARAMS=dmic_default,\
+PREPROCESS_PLUGINS=nhlt,NHLT_BIN=sof-lnl-cs42l43-l0-cs35l56-l23-4ch.bin"
+
 "cavs-sdw\;sof-lnl-cs42l43-l2-cs35l56x6-l13\;NUM_SDW_AMP_LINKS=3,SDW_DMIC=1,\
 SDW_AMP_FEEDBACK=false,SDW_SPK_STREAM=Playback-SmartAmp,SDW_DMIC_STREAM=Capture-SmartMic,\
 SDW_JACK_OUT_STREAM=Playback-SimpleJack,SDW_JACK_IN_STREAM=Capture-SimpleJack"
```

@ujfalusi I modified to `DMIC0_ID=5,DMIC1_ID=6,HDMI1_ID=7,HDMI2_ID=8,HDMI3_ID=9` as the dmesg shows. Now I can confirm that both playback (speaker/HDMI) and capture are working.

ujfalusi, 2025-03-27:
It might be that the same quirk is needed for S16 as it was needed for S14 in kernel?
https://github.com/thesofproject/linux/commit/0c2bf750fcdd6df578e52c0603081a375cfcf1f1

@rfvirgil, do you know?

simontrimmer, 2025-03-27:
@ujfalusi - As i think @rfvirgil may have his hands full, I think it probably does need that quirk

ujfalusi, 2025-03-28:
@PisonJay, can you provide the output of `alsa-info.sh` ?

ujfalusi, 2025-03-28:
https://github.com/thesofproject/sof/pull/9934, which will need similar kernel patch as S14 )https://github.com/thesofproject/linux/commit/0c2bf750fcdd6df578e52c0603081a375cfcf1f1), for which we need to alsa-info to get the required PCI ID.

PisonJay, 2025-03-28:
@ujfalusi, here you are. [alsa-info.txt](https://github.com/user-attachments/files/19500806/alsa-info.txt)

ujfalusi, 2025-03-28:
@PisonJay, thank you. The kernel patch is: https://github.com/thesofproject/linux/pull/5377

kv2019i, 2025-03-31:
The missing topology file is now released in https://github.com/thesofproject/sof-bin/releases/tag/v2025.01.1

closed 2025-04-01 by abonislawski

ujfalusi, 2025-04-04:
@PisonJay, I have sent the kernel patch for 6.15 and if it is taken, it is going to be backported to 6.14.
https://lore.kernel.org/linux-sound/20250404133213.4658-1-peter.ujfalusi@linux.intel.com/

Let's hope that it makes it through soon and lands in Arch.