thesofproject/sof #9940: [BUG] Zephyr panic excause 63 in nocodec capture test check-capture-50rounds

kv2019i, 2025-03-31:
**Describe the bug**

DSP panic seen in daily tests on Intel PTL platform running check-capture-50rounds. First seen on daily test of 21st Feb with following dump in FW trace:

```
[00:01:42.280,021]  os: print_fatal_exception:  ** FATAL EXCEPTION
[00:01:42.280,028]  os: print_fatal_exception:  ** CPU 2 EXCCAUSE 63 (zephyr exception)
[00:01:42.280,031]  os: print_fatal_exception:  **  PC 0xa0030577 VADDR 0xa005e9a0
[00:01:42.280,035]  os: print_fatal_exception:  **  PS 0x60023
[00:01:42.280,036]  os: print_fatal_exception:  **    (INTLEVEL:3 EXCM: 0 UM:1 RING:0 WOE:1 OWB:0 CALLINC:2)
[00:01:42.280,040]  os: xtensa_dump_stack:  **  A0 0xa005cbc0  SP 0xa00b01d0  A2 0x4  A3 0xa00b01e0
[00:01:42.280,043]  os: xtensa_dump_stack:  **  A4 0xa00b01c0  A5 0x4  A6 0x47f  A7 0x4
[00:01:42.280,046]  os: xtensa_dump_stack:  **  A8 0xa005cb0a  A9 0xa00b0160 A10 0xa0083d85 A11 0xa00b01e0
[00:01:42.280,050]  os: xtensa_dump_stack:  ** A12 0xa00b01c0 A13 0x4 A14 0x1 A15 0x480
[00:01:42.280,051]  os: xtensa_dump_stack:  ** LBEG 0xa00388a5 LEND 0xa00388b4 LCOUNT 0xa005d510
[00:01:42.280,055]  os: xtensa_dump_stack:  ** SAR 0x1
[00:01:42.280,058]  os: xtensa_dump_stack:  **  THREADPTR 0xfffffff4

--
Backtrace:0xa0030574:0xa00b01d0  |<-CORRUPTED
```

**To Reproduce**
sof-test check-capture-50rounds

**Reproduction Rate**
open -- in multiple daily test plans since 21st Feb

**Expected behavior**
No DSP panics

**Impact**
Loss of audio functionality

**Environment**
1) SOF main , latest sof-dev kernel
2) Name of the topology file
    * Topology: sof-ptl-nocodec.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: ptl

kv2019i, 2025-03-31:
PRs merged before 21st: https://github.com/thesofproject/sof/pull/9850 (21st), https://github.com/thesofproject/sof/pull/9906 (20th), https://github.com/thesofproject/sof/pull/9848 (20th), https://github.com/thesofproject/sof/pull/9900 (21st)

kv2019i, 2025-03-31:
Not seen in pull-requests tests as the panic usually triggers after at least 20 iterations of the check-capture test (i.e. PR CI test is too short to capture this fail).

lyakh, 2025-04-02:
#9942 should fix this, merged, next daily test will tell

lyakh, 2025-04-22:
Haven't been able to find any more of these failures since #9946 was merged on Apr. 2nd. Propose to close

wszypelt, 2025-04-25:
@kv2019i request for verification and close

kv2019i, 2025-04-25:
Aa, sorry, I tried to close this already before, but hit a github glitch. So based on PR/daily test results, this issue is fixed and can be closed.

closed 2025-04-25 by kv2019i