thesofproject/sof #9973: [BUG] No sound output from built-in speakers  on an Intel Tiger Lake-LP system with an ESSX8336 audio codec

andycandy, 2025-04-20:
**Describe the bug**
No sound output from built-in speakers or wired headphone jack on an Intel Tiger Lake-LP system with an ESSX8336 audio codec. The sound card is detected by the system (ALSA/PipeWire), but the SOF kernel driver fails to initialize the hardware streams during boot, resulting in no audio playback from the built-in audio device. Bluetooth audio works correctly.

What have you tried to diagnose or workaround this issue?

- Checked volume levels and output device selection in GNOME Settings (Correct device "Speakers - Tiger Lake-LP Smart Sound Technology Audio Controller" is shown and selected, volume is up, not muted in GUI).
- Checked alsamixer for the sof-essx8336 card (Speaker and Headphone channels are visible and unmuted - OO).
- Tested wired headphones (No sound).
- Tested Bluetooth earbuds (Sound works correctly).
- Restarted PipeWire services (systemctl --user restart pipewire pipewire-pulse wireplumber).
- Reinstalled sof-firmware, alsa-sof-firmware, linux-firmware, and alsa-ucm-conf packages using dnf.
- Tried adding the kernel parameter snd_intel_dspcfg.dsp_driver=2 to GRUB_CMDLINE_LINUX (This stopped the -5 errors in dmesg but caused the system to load the incorrect snd_soc_avs driver, resulting in only "Dummy Output". This parameter was removed).
- Checked lspci to verify the loaded kernel driver.
- Checked dmesg for errors related to sof, essx, snd, audio.

Please also read https://thesofproject.github.io/latest/contribute/process/bug-tracking.html for further information on submitting bugs.
[Acknowledged]

**To Reproduce**
Steps to reproduce the behavior: (e.g. list commands or actions used to reproduce the bug)

1. Boot the system with the specified kernel and configuration.
2. Attempt to play audio through the built-in speakers or wired headphone jack (e.g., using the sound test in GNOME Settings, playing a video in a browser).

### Reproduction Rate
**How often does the issue happen ? i.e. 1/10 (once in ten attempts), 1/1000 or all the time.**

   All the time. The issue occurs on every boot with the default kernel parameters (without snd_intel_dspcfg.dsp_driver=2).

**Does the reproduction rate vary with any other configuration or user action, if so please describe and show the new reproduction rate.**

-  Using snd_intel_dspcfg.dsp_driver=2 stops the -5 EIO errors, but the audio device still doesn't work (only "Dummy Output" is available as the wrong driver is loaded).
- Using Bluetooth audio works reliably.

Expected sound output from the built-in speakers and wired headphone jack when selected as the output device.

**Impact**
Showstopper for using the built-in audio hardware. Requires using Bluetooth audio or external devices for sound. Significant annoyance.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {This information is not readily available for a standard Fedora package, usually provided by developers debugging. You can state the package version instead}
        *   Kernel Package: `kernel.x86_64 6.14.0-63.fc42`
        *   Kernel Package (also tested): `kernel.x86_64 6.14.2-300.fc42`
    * SOF:
        *   SOF Firmware Package: `sof-firmware` (latest) 
        *   ALSA SOF Firmware Package: `alsa-sof-firmware 2025.01-1.fc42` 
2) Name of the topology file
    * Topology: Cannot determine exact filename. The codec is ESSX8336 on Intel SOF.
3) Name of the platform(s) on which the bug is observed.
    * Platform: Intel Tiger Lake-LP (TGL-LP)
        *   PCI ID: `8086:a0c8`
        *   Subsystem ID: `2782:1a32` (Emdoor Digital Technology Co., Ltd Device)
        *   Codec: ESSX8336 (`sof-essx8336`)
        *   Laptop Model: Infinix INBook X2 Inbook i3 1115G4 

Output of `lspci -knn | grep -iA3 audio`:
```
00:1f.3 Multimedia audio controller [0401]: Intel Corporation Tiger Lake-LP Smart Sound Technology Audio Controller [8086:a0c8] (rev 30)
	DeviceName: Onboard - Sound
	Subsystem: Emdoor Digital Technology Co., Ltd Device [2782:1a32]
	Kernel driver in use: sof-audio-pci-intel-tgl
	Kernel modules: snd_hda_intel, snd_soc_avs, snd_sof_pci_intel_tgl
00:1f.4 SMBus [0c05]: Intel Corporation Tiger Lake-LP SMBus Controller [8086:a0a3] (rev 30)
	DeviceName: Onboard - Other
```

Output of `sudo dmesg | grep -i 'sof\|essx' | grep -i 'error\|fail'`:
```
[   XX.XXXXXX] sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx error for 0x60010000 (msg/reply size: 108/20): -5
[   XX.XXXXXX] sof-audio-pci-intel-tgl 0000:00:1f.3: HW params ipc failed for stream 1
[   XX.XXXXXX] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_pcm_component_hw_params on 0000:00:1f.3: -5
... (Many repetitions of these errors) ...
```

Output of `cat /proc/asound/card*/codec* | grep Codec`:
```
Codec: ESSX8336
```

Output of `uname -r`:
```
6.14.0-63.fc42.x86_64
```

Please also include the relevant sections from the firmware log and kernel log in the report (and attach the full logs for complete reference). Kernel log is taken from *dmesg* and firmware log from *sof-logger*. See https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html

abonislawski, 2025-04-22:
@andycandy could you please share full dmesg and fw logs?

kv2019i, 2025-04-22:
@andycandy Can you provide output of alsa-info (ref https://thesofproject.github.io/latest/getting_started/intel_debug/suggestions.html )?

andycandy, 2025-04-22:
<html><head></head><body>
<p>Thanks for the follow-up. I've made partial progress toward getting the internal audio working using a combination of custom files and kernel parameters. However, there is currently a trade-off: I can get <strong>stereo speaker output without a functional microphone</strong>, or <strong>mono speaker output with a working microphone</strong>—but not both at the same time.</p>
<h3><strong>Steps Taken &amp; Results:</strong></h3>
<h4>1. <strong>Custom Kernel Files + Quirk Parameter (Enables Stereo Output):</strong></h4>
<ul>
<li>
<p>Cloned <code inline="">https://github.com/yangxiaohua2009/custom-kernel.git</code> and executed the included <code inline="">make.sh</code> to copy the necessary <code inline="">firmware</code>, <code inline="">topology</code>, and <code inline="">UCM</code> files.</p>
</li>
<li>
<p>Added <code inline="">snd_soc_sof_es8336.quirk=128</code> to GRUB.</p>
</li>
<li>
<p><strong>Result:</strong> After reboot:</p>
<ul>
<li>
<p><strong>Stereo speaker output works</strong>, but required manually boosting the DAC level in <code inline="">alsamixer</code> to hear sound.</p>
</li>
<li>
<p><strong>Microphone is not detected</strong> in GNOME Settings or applications.</p>
</li>
<li>
<p><code inline="">dmesg</code> still shows persistent <code inline="">-5</code> EIO errors (<code inline="">ipc tx error</code>, <code inline="">HW params ipc failed</code>).</p>
</li>
</ul>
</li>
</ul>
<h4>2. <strong>Force DMIC Detection &amp; ALSA Capture Test:</strong></h4>
<ul>
<li>
<p>Created <code inline="">/etc/modprobe.d/alsa-dmic-fix.conf</code> with <code inline="">options snd_sof_intel_hda_common dmic_num=2</code>.</p>
</li>
<li>
<p>Rebooted (with <code inline="">quirk=128</code> still active).</p>
</li>
<li>
<p>Verified capture device via <code inline="">arecord -l</code>:<br>
<code inline="">card 0: sofessx8336 [sof-essx8336], device 0: ES8336 (*) []</code>.</p>
</li>
<li>
<p><strong>Result:</strong> <code inline="">arecord -D hw:0,0 -c2 -r48000 -f S16_LE -d 10 test.wav</code> successfully captured audio.<br>
This confirmed the <strong>mic hardware and low-level ALSA path are functional</strong>.</p>
</li>
</ul>
<h4>3. <strong>Editing UCM Profile (No App-Level Mic Detection):</strong></h4>
<ul>
<li>
<p>Modified <code inline="">/usr/share/alsa/ucm2/sof-essx8336/HiFi.conf</code>, particularly:</p>
<ul>
<li>
<p>In <code inline="">SectionDevice."Mic"</code>, set:</p>
<ul>
<li>
<p><code inline="">CapturePCM "hw:0,0"</code></p>
</li>
<li>
<p><code inline="">CaptureChannels "2"</code></p>
</li>
</ul>
</li>
<li>
<p>Also tested by commenting out <code inline="">EnableSequence</code> block.</p>
</li>
</ul>
</li>
<li>
<p>Restarted PipeWire/WirePlumber.</p>
</li>
<li>
<p><strong>Result:</strong></p>
<ul>
<li>
<p>GNOME and most applications <strong>still do not detect the microphone</strong> under the "Stereo Output + Stereo Input" profile.</p>
</li>
<li>
<p><code inline="">pavucontrol</code> showed a new <strong>"Headset Microphone (unplugged)"</strong> source, but it remained <strong>non-functional</strong>.</p>
</li>
</ul>
</li>
</ul>
<h4>4. <strong>Switching to "Pro Audio" Profile (Mic Works, Stereo Lost):</strong></h4>
<ul>
<li>
<p>In <code inline="">pavucontrol</code> &gt; Configuration tab, switched profile to <strong>"Pro Audio"</strong>.</p>
</li>
<li>
<p><strong>Result:</strong></p>
<ul>
<li>
<p>The <strong>microphone appears and works</strong> in GNOME Settings and applications.</p>
</li>
<li>
<p>However, <strong>speaker output becomes mono</strong>.</p>
</li>
<li>
<p>Reverting to <strong>"Stereo Output + Stereo Input"</strong> brings back stereo speakers but makes the mic unusable again.</p>
</li>
</ul>
</li>
</ul>
<hr>
<h3><strong>Current State:</strong></h3>
<p>With <code inline="">quirk=128</code>, <code inline="">dmic_num=2</code>, and the custom UCM/firmware setup:</p>

Profile | Speakers | Microphone
-- | -- | --
Stereo Output + Stereo Input | ✅ Stereo | ❌ Not detected
Pro Audio | ⚠️ Mono only | ✅ Working

<p>It seems the <strong>"HiFi" UCM profile fails to expose the <code inline="">hw:0,0</code> capture device correctly</strong>, even after edits. Meanwhile, the <strong>"Pro Audio" profile bypasses UCM mic mapping issues</strong>, but doesn’t use stereo speaker output routing.</p>
<p>Persistent <code inline="">-5</code> EIO errors in <code inline="">dmesg</code> may be connected to this incomplete profile/initialization state.</p>
<hr>
<h3><strong>Diagnostics (Current State = Pro Audio Profile Active):</strong></h3>
<p>Mic works, speakers are mono, <code inline="">quirk=128</code> and <code inline="">dmic_num=2</code> active.<br>
<strong>Alsa-info dump:</strong><br>
🔗 <a href="http://alsa-project.org/db/?f=33fb495347350021faa0a870501b1444e60c9440">http://alsa-project.org/db/?f=33fb495347350021faa0a870501b1444e60c9440</a></p>
<hr>
<h3><strong>Request for Guidance:</strong></h3>
<ul>
<li>
<p>How can I modify the <strong>"HiFi.conf" UCM profile</strong> (used by the "Stereo Output + Stereo Input" profile) to correctly expose and route the <code inline="">hw:0,0</code> mic while preserving stereo speaker output?</p>
</li>
<li>
<p>Alternatively, is it possible to <strong>enable stereo output under "Pro Audio"</strong> or another profile that already has the mic working?</p>
</li>
</ul>
<p>Happy to provide full <code inline="">dmesg</code> and firmware logs if needed.</p>
<p>Thanks again!</p>

Abhinav5383, 2025-04-29:
I also have the same laptop, I get audio from the speaker after installing the `sof-firmware` package
but its so low at 100% volume that I can't hear it clearly even after sticking my ear into the speakers

(I do have audio on any externally connected device like wired earphones and bluetooth earphones)

For me though, switching to `Pro audio` profile stops the audio completely, the mic works with neither one

andycandy, 2025-04-29:
I can help you out, try figuring out which mic do you have working on lower level, trying running the code in terminal to get the hdd: <values> those would be your mic ports and you'll have to manually add them to sof conf, its do-able

Abhinav5383, 2025-04-30:
@andycandy I tried the profile switching thing through pavucontrol again after reinstalling `sof-firmware` and turns out the mic does indeed work on `Pro audio` profile, the speaker volume is still lower than the sound of someone whispering from miles away

so seems like mine is now in the same condition as yours

- Stereo Output + Stereo Input : Only speaker works, mic is not even listed 
- Pro audio : Mic works , so does the speaker 

*I can't determine if the sound coming from the speaker is stereo or not because It's so low I can't hear both speakers simultaneously 

One more thing I noticed is that even though the speaker is producing like 0.01% volume, the sound cracks so much that it seems like the volume's been scaled down from 100%(or probably more) to whatever it's producing

andycandy, 2025-04-30:
For lower audio you have to just increase it from alsamixer blud

Abhinav5383, 2025-04-30:
I don't use alsa, it's all pipewire

Abhinav5383, 2025-04-30:
@andycandy I followed the steps you took to fix your speaker (though I didn't use the bash script from the custom-kernel repo because the files it was copying are already there) and set `snd_soc_sof_es8336.quirk=128`
the sound was still low, so increased the DAC level to around 70 (sounds starts cracking beyond this)
and I've got usable sound from the speakers now

Regarding the mic, idk what's happening
pavucontrol shows `Headset Microphone (unplugged)` but captures the mic correctly
any other application can't detect the mic

Switching to the `Pro audio` profile makes the mic detectable but speaker turns mono
so basically same as you

thanks for the help btw :)

andycandy, 2025-04-30:
Yeah, I don't know how to fix that mono audio fam, but anyways. Glad to help!

kv2019i, 2025-05-06:
Let's move this to Linux driver component as the errors are more related to drivers than the FW. The hw_params errors are caused by Pipewire/pulsewire doing pararameter discovery at startup (trying all possible PCM formats to see which are supported).