thesofproject/sof #9983: [FEATURE] remove XTOS support from SOF mainline

kv2019i, 2025-05-02:
**Is your feature request related to a problem? Please describe.**
Since 2022, SOF has been moving to Zephyr as its primary RTOS with work tracked in https://github.com/thesofproject/sof/issues/5794
Most SOF users have either completed the transition to Zephyr, or are in progress of adding support for Zephyr for their platforms.

This ticket is for removal of XTOS support from SOF main, leaving XTOS support only supported in release stable branches. There is a set of cleanups we can proceed with, once XTOS support is removed, so having a ticket to mark as a dependency is useful for tracking.

**Describe the solution you'd like**
XTOS support removal from "main" branch

**Describe alternatives you've considered**
-

kv2019i, 2025-05-07:
@lgirdwood opened discussion at https://github.com/orgs/thesofproject/discussions/9993

lgirdwood, 2025-05-07:
@thesofproject/amd @thesofproject/nxp @thesofproject/intel @thesofproject/google @thesofproject/mediatek pls comment/review and provide any feedback.

bhiregoudar, 2025-05-07:
> [@thesofproject/amd](https://github.com/orgs/thesofproject/teams/amd) [@thesofproject/nxp](https://github.com/orgs/thesofproject/teams/nxp) [...]
@lgirdwood  : We are in transition to Zephyr. DMA scheduling is removed in Zephyr and we have an issue with timer scheduler working, So, till we resolve timer based scheduler issue we need to support XTOS on AMD platforms.

lgirdwood, 2025-05-07:
> [@thesofproject/amd](https://github.com/orgs/thesofproject/teams/amd) [@thesofproject/nxp](https://github.com/orgs/thesofproject/teams/nxp) [...]

@bhiregoudar ok, do you have a schedule for fixing timer scheduling on AMD ? 

Fwiw, we found DMA based scheduling would not scale beyond a few streams as stream start ordering meant DMA IRQs would fire in different sequences/times and could cause a lot of thrash and glitches...

Timer based scheduling works best when the DSP is clock master for the codecs (otherwise you need to add the ASRC), as all pipelines are executed from a single synchronous tick in a deterministic order. Feel free to create an issue and it should gather comments from others who have transitioned to timers for debug.

lgirdwood, 2025-05-08:
@andrew-mtk pls reply with any comments for Mediatek team. @andyross feedback welcome too.

lgirdwood, 2025-05-09:
> IMO all good from NXP's side. Although [@lgirdwood](https://github.com/lgirdwood) already mentioned this I want to stress out the fact that all IMX platforms are still using the XTOS IPC driver so we need to watch out for that.

Yeah - that's something that @kv2019i is working on with Zephyr folks. We also need Zephyr clock driver too. We would need to keep the legacy drivers until Zephyr replacements were upstream.

kv2019i, 2025-05-09:
> Yeah - that's something that [@kv2019i](https://github.com/kv2019i) is working on with Zephyr folks. We also need Zephyr clock driver too. We would need to keep the legacy drivers until Zephyr [...]

- IPC work tracked in https://github.com/thesofproject/sof/issues/9697
- Clock work tracked in https://github.com/thesofproject/sof/issues/9541 (only Intel uses the more complex clock change interface, so this should be less of a blocker as there's no XTOS user left anyways)

lgirdwood, 2025-05-19:
> [@bhiregoudar](https://github.com/bhiregoudar) ok, do you have a schedule for fixing timer scheduling on AMD ?
>
> Fwiw, we found DMA based scheduling would not scale beyond a few streams as stream [...]

@bhiregoudar any update here ?

lgirdwood, 2025-05-19:
> [@andrew-mtk](https://github.com/andrew-mtk) pls reply with any comments for Mediatek team. [@andyross](https://github.com/andyross) feedback welcome too.

@andrew-mtk @andyross ping.

bhiregoudar, 2025-05-20:
> [@bhiregoudar](https://github.com/bhiregoudar) any update here ?

We have started working on timer scheduler. One question is that, if XTOS is removed, can the already existing platforms continue with XTOS? What about the ToT driver support to the older platforms which uses XTOS? any dependency?

kv2019i, 2025-07-03:
@lgirdwood You added a list of tasks including removal of SOF abstractions for dma/trace/memory. Do you want to track them here, or in https://github.com/thesofproject/sof/issues/5794 . I don't think we need to track in both places. XTOS is now removed, and remaining work is to reduce abstractions on SOF side.

johnylin76, 2025-07-05:
> > > > > > [@thesofproject/amd](https://github.com/orgs/thesofproject/teams/amd) [@thesofproject/nxp](https://github.com/orgs/thesofproject/teams/nxp) [@thesofproject/intel](https://github.com/orgs/thesofproject/teams/intel) [@thesofproject/google](https://github.com/orgs/thesofproject/teams/google) [@thesofproject/mediatek](https://github.com/orgs/thesofproject/teams/mediatek) pls comment/review and provide any feedback.
> > > > > > [@lgirdwood](https://github.com/lgirdwood)  : We are in transition to Zephyr. DMA scheduling is removed in Zephyr and we have an issue with timer scheduler working, So, till we resolve timer based scheduler issue we need to support XTOS on AMD platforms.
> > >
> > > ...
> > 
> > We have started working on timer scheduler. One question is that, if XTOS is removed, can the already existing platforms continue with XTOS? What about the ToT driver support to the older platforms which uses XTOS? any dependency?
> 
> Yes, already existing platforms can continue with SOF v2.13 release branch (Intel does this with SOF v2.2 release branch for older IPC3 platforms).
> 
> The main branch is for future/current platforms where we want to accelerate velocity and save effort using Zephyr RTOS.

Could I regard this as the SOF mainline will no more support XTOS after v2.13. In other words, if one on main branch would like to build XTOS, it should base on the commit no later than v2.13 tag.

By the way, is the use of IPC4 bound together with Zephyr? From my understanding, they should be independent which means Zephyr build + IPC3 should be expected to work as well, is that correct? Is there a plan to remove IPC3?

lgirdwood, 2025-09-30:
> Updated the checkbox list for current work done in v2.14. Basicly XTOS platform support is completely removed, but there v2.14 keeps ability to use XTOS drivers in Zephyr and there are many drivers in the tree that use this. We have moved to many native APIs, but some SOF side abstractions remains.

@thesofproject/amd @thesofproject/nxp @thesofproject/mediatek whats your status with Zephyr native drivers ? Are you able to start deleting SOF FW drivers in SOF codebase ?

bhiregoudar, 2025-10-07:
> [@thesofproject/amd](https://github.com/orgs/thesofproject/teams/amd) [@thesofproject/nxp](https://github.com/orgs/thesofproject/teams/nxp) [...]

We have moved to Native Zephyr drivers for DAI, DMA and interrupt.  We haven't up streamed the code yet. We are in process of upstream. We need to look into other drivers like mailbaox drivers to move to native driver.

lyakh, 2026-01-30:
@thesofproject/amd one of the features that are being deprecated here is `CONFIG_TRACE` and the only platform enabling it is vangogh: https://github.com/thesofproject/sof/blob/0187fc51ef0356f2615b5d330638a267f1675c43/src/arch/xtensa/configs/vangogh_defconfig#L24 Is it indeed used there on the "main" branch?