thesofproject/sof #9997: [BUG] IMX: Topology build fails after recent change

dbaluta, 2025-05-09:
**Describe the bug**

Building topology now fails after commit https://github.com/thesofproject/sof/commit/47dd8d33fc9ead3ee2d3e378a9ad93e889d41e68

Our environment adds the alsatplg using the PATH but the commit mentioned above uses an explicit alsatplg. This breaks the build.

**To Reproduce**

./scripts/build-tools.sh 

**Reproduction Rate**

100%

**Expected behavior**

build shouldn't break

**Impact**
Building SOF binaries fails.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: doesnt matter
    * SOF: 137c9bed058662815f66355b52705c6bfc40d4b4
     
2) Name of the topology file
    * Topology: topology doesnt build
3) Name of the platform(s) on which the bug is observed.
    * Platform: imx8m

dbaluta, 2025-05-09:
@lgirdwood can you add the alsatplg in the PATH variable? Is it OK to revert this change?

lgirdwood, 2025-05-09:
> [@lgirdwood](https://github.com/lgirdwood) can you add the alsatplg in the PATH variable? Is it OK to revert this change?

@dbaluta do you mean adding the local ALSA binaries to the users $PATH ? This could conflict with system ALSA/audio. Can you share the error you are seeing. Running the script should build all topologies.

dbaluta, 2025-05-13:
```
  /home/bamboo/sof_dir/sof/tools/../../tools/bin/alsatplg --version returned
  status: No such file or directory,
Call Stack (most recent call first):
  topology/CMakeLists.txt:34 (alsatplg_version)
  CMake Warning at topology/CMakeLists.txt:36 (message):
  /home/bamboo/sof_dir/sof/tools/../../tools/bin/alsatplg failed: No such file or directory; all topologies skipped

 ```

And it expected because SOF scripts expect the alsatplg to be installed in a custom directory ``  /home/bamboo/sof_dir/sof/tools/../../tools/bin/alsatplg`` but on our machine the alsatplg local compiled version is installed in a different directory.

kv2019i, 2025-05-13:
@dbaluta We discussed these concerns in #9870 review and e.g. about the location in https://github.com/thesofproject/sof/pull/9870#discussion_r2058571180 . SOF_WORKSPACE (I admit, I didn't realize we had this in our original public docs) can be used to change adjust the install location, but there's no option to use system tplg.

kv2019i, 2025-05-16:
@dbaluta Any takes how we want to proceed? I'd need to backport Liam's PR to stable-v2.2 to unblock the CI, but yeah, do we stick this approach on main. 

Mandating local ALSA installs does have the nice property that we can introduce topology features changes in SOF pull-requests, adding the alsa-lib/utils changes at the same time as the changes to topologies, and have them tested at same time. Now we frequently have issues with people using old alsa-lib/utils (either omission or mistake in tool install), resulting in buggy topologies binaries.

lgirdwood, 2025-05-16:
> I don't have a strong opinion on this, but I would like to have an option to use a different location OR at least use the system alsatplg. We the current implementation is no longer possible, user is forced to install the alsa tools in a exact location, otherwise the build will fail.

@dbaluta So I guess this also is needed by Yocto? as it 's ALSA recipe is probably very much more bleeding edge than say Fedora or Ubuntu (which is why building with system ALSA does not work). No objections though if a command line arg is needed to override ALSA $PATH

dbaluta, 2025-05-16:
@kv2019i @lgirdwood we fixed this internally in Yocto for now, we can close this and will come back maybe with an improvement if necessary later.

closed 2025-05-16 by dbaluta