thesofproject/sof #10014: [BUG] IMX95 build fails with May15 Zephyr main

kv2019i, 2025-05-16:
**Describe the bug**
IMX95 ARM build is failing on latest Zephyr main: 

https://github.com/thesofproject/sof/actions/runs/15046448636/job/42340690787?pr=10011
```/home/cwd_user/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DCPU_MIMX9596AVZXN_cm7 -DFSL_SDK_DISABLE_DRIVER_CLOCK_CONTROL=1 -DKERNEL -DK_HEAP_MEM_POOL_SIZE=2048 -DNDEBUG -D__ZEPHYR__=1 -I/zep_workspace/zephyr/kernel/include -I/zep_workspace/zephyr/arch/arm/include -I/zep_workspace/build-imx95/zephyr/include/generated/zephyr -I/zep_workspace/zephyr/include -I/zep_workspace/build-imx95/zephyr/include/generated -I/zep_workspace/zephyr/soc/nxp/imx -I/zep_workspace/zephyr/soc/nxp/imx/imx9/imx95/. -I/zep_workspace/zephyr/soc/nxp/imx/imx9/imx95/m7/. -I/zep_workspace/zephyr/soc/nxp/imx/. -I/zep_workspace/zephyr/soc/nxp/imx/imx9 -I/zep_workspace/zephyr/soc/nxp/imx/imx9/include -I/zep_workspace/sof/zephyr/include -I/zep_workspace/zephyr/arch/xtensa/include -I/zep_workspace/sof/src/platform/imx95/include -I/zep_workspace/modules/hal/nxp/mcux/mcux-sdk/devices/MIMX9596 -I/zep_workspace/modules/hal/nxp/mcux/mcux-sdk/devices/MIMX9596/drivers -I/zep_workspace/modules/hal/nxp/mcux/mcux-sdk/drivers/common 
In file included from /zep_workspace/zephyr/include/zephyr/arch/arm/asm_inline.h:18,
                 from /zep_workspace/zephyr/include/zephyr/arch/arm/arch.h:30,
                 from /zep_workspace/zephyr/include/zephyr/arch/cpu.h:19,
                 from /zep_workspace/zephyr/include/zephyr/kernel_includes.h:36,
                 from /zep_workspace/zephyr/include/zephyr/kernel.h:17,
                 from /zep_workspace/zephyr/arch/arm/core/offsets/offsets_aarch32.c:28,
                 from /zep_workspace/zephyr/arch/arm/core/offsets/offsets.c:9:
/zep_workspace/zephyr/include/zephyr/arch/arm/asm_inline_gcc.h:24:10: fatal error: cmsis_core.h: No such file or directory
   24 | #include <cmsis_core.h>
```

kv2019i, 2025-05-16:
Seen in #10011  and #10007

kv2019i, 2025-05-16:
@dbaluta can you check?

dbaluta, 2025-05-16:
Thanks for report! 

Cc: @LaurentiuM1234

closed 2025-05-19 by dbaluta