thesofproject/sof #10029: [BUG] Firmware download failed

dombong, 2025-05-24:
**Describe the bug**

I don't get input/output devices

**To Reproduce**
On boot

**Reproduction Rate**
All the time

**Expected behavior**
I get input and output device

**Impact**
Annoyance since bluetooth audio still work

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 788aadc8f73d93908cf8545b2fc770fb6580a3f2 (20250509)
    * SOF: f46921ffc67107f1c9e67c22307d63d370f18fcb (v2.12.1)
2) Name of the topology file
    * Topology:  intel/sof-ipc4-tplg/sof-lnl-cs42l43-l0-cs35l56-l23-2ch.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: NixOS 25.11

**Screenshots or console output**
Sorry I'm not sure how to run `sof-logger` or `smex`

[dmesg.log](https://github.com/user-attachments/files/20420868/dmesg.log)

In dmesg
```
[    0.041999] software IO TLB: area num 8.
[    0.250800] kobject: kobject_add_internal failed for mipi-sdca-control-0x2-subproperties with -EEXIST, don't try to register things with the same name in the same directory.
[    0.250831] kobject: kobject_add_internal failed for mipi-sdca-control-0x2-subproperties with -EEXIST, don't try to register things with the same name in the same directory.
[    0.539699] PCI-DMA: Using software bounce buffering for IO (SWIOTLB)
[    0.539700] software IO TLB: mapped [mem 0x000000004fc16000-0x0000000053c16000] (64MB)
[    3.204651] intel-hid INTC107B:00: failed to get button capability
[    3.529693] asus_wmi: fan_curve_get_factory_default (0x00110024) failed: -19
[    3.529912] asus_wmi: fan_curve_get_factory_default (0x00110025) failed: -19
[    3.530106] asus_wmi: fan_curve_get_factory_default (0x00110032) failed: -19
[    3.659023] sof-audio-pci-intel-lnl 0000:00:1f.3: enabling device (0000 -> 0002)
[    3.659128] sof-audio-pci-intel-lnl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[    3.915317] sof-audio-pci-intel-lnl 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [xe])
[    3.922249] sof-audio-pci-intel-lnl 0000:00:1f.3: use msi interrupt mode
[    3.940195] sof-audio-pci-intel-lnl 0000:00:1f.3: hda codecs found, mask 4
[    3.940205] sof-audio-pci-intel-lnl 0000:00:1f.3: NHLT device BT(0) detected, ssp_mask 0x4
[    3.940207] sof-audio-pci-intel-lnl 0000:00:1f.3: BT link detected in NHLT tables: 0x4
[    3.940210] sof-audio-pci-intel-lnl 0000:00:1f.3: DMICs detected in NHLT tables: 2
[    3.944937] sof-audio-pci-intel-lnl 0000:00:1f.3: Firmware paths/files for ipc type 1:
[    3.944948] sof-audio-pci-intel-lnl 0000:00:1f.3:  Firmware file:     intel/sof-ipc4/lnl/sof-lnl.ri
[    3.944949] sof-audio-pci-intel-lnl 0000:00:1f.3:  Firmware lib path: intel/sof-ipc4-lib/lnl
[    3.944950] sof-audio-pci-intel-lnl 0000:00:1f.3:  Topology file:     intel/sof-ipc4-tplg/sof-lnl-cs42l43-l0-cs35l56-l23-2ch.tplg
[    3.945493] sof-audio-pci-intel-lnl 0000:00:1f.3: Loaded firmware library: ADSPFW, version: 2.12.0.1
[    4.095299] intel_ish_ipc 0000:00:12.0: ISH loader: cmd 2 failed 10
[    4.146738] intel_ish_ipc 0000:00:12.0: ISH loader: cmd 2 failed 10
[    4.187325] intel_ish_ipc 0000:00:12.0: ISH loader: cmd 2 failed 10
[    6.949243] sof-audio-pci-intel-lnl 0000:00:1f.3: hda_cl_copy_fw: timeout with rom_status_reg (0x160200) read
[    6.949766] sof-audio-pci-intel-lnl 0000:00:1f.3: ------------[ DSP dump start ]------------
[    6.950262] sof-audio-pci-intel-lnl 0000:00:1f.3: Firmware download failed
[    6.950757] sof-audio-pci-intel-lnl 0000:00:1f.3: fw_state: SOF_FW_BOOT_IN_PROGRESS (3)
[    6.951258] sof-audio-pci-intel-lnl 0000:00:1f.3: 0xd000000c: module: ROM_EXT, state: VALIDATE_PUB_KEY, not running
[    6.951750] sof-audio-pci-intel-lnl 0000:00:1f.3: error code: 0x97 (unknown)
[    6.952268] sof-audio-pci-intel-lnl 0000:00:1f.3: ------------[ DSP dump end ]------------
[    6.952770] sof-audio-pci-intel-lnl 0000:00:1f.3: Failed to start DSP
[    6.953280] sof-audio-pci-intel-lnl 0000:00:1f.3: error: failed to boot DSP firmware -110
[    7.466701] soundwire sdw-master-0-0: trf on Slave 6 failed:-5 write addr 9008 count 0
[    7.971876] soundwire sdw-master-0-0: trf on Slave 6 failed:-5 read addr 9008 count 0
[    8.482156] soundwire sdw-master-0-0: trf on Slave 6 failed:-5 write addr b0dc count 0
[    8.986256] soundwire sdw-master-0-0: trf on Slave 6 failed:-5 read addr b0dc count 0
...
[   83.586494] sof-audio-pci-intel-lnl 0000:00:1f.3: error: sof_probe_work failed err: -110
```

lgirdwood, 2025-05-26:
@ujfalusi @bardliao fyi

lgirdwood, 2025-05-26:
Similar to #10025 - known validated FW versions failing to boot, could a distro be missing some kernel patches.

ujfalusi, 2025-05-27:
@dombong, if you have installed the sof-bin by hand, make sure that you have copied the entire sof-ipc4 directory as it is to /lib/firmware/intel/
and make sure you use the latest release, which is atm: https://github.com/thesofproject/sof-bin/releases/tag/v2025.01.1

bardliao, 2025-05-27:
@dombong May I know what device are you using? And where do you get the SOF firmware binary?

dombong, 2025-06-28:
I managed to upgrade the BIOS to latest version offered by ASUS and it's the same problem, sadly.

I double check that MEI/CSM is running AFAIK
```
$ lsmod | rg mei  
mei_gsc_proxy          16384  0
mei_me                 57344  1
mei                   196608  3 mei_gsc_proxy,mei_me
$ sudo cat /sys/class/mei/mei0/fw_status
A0000245
89210500
00000020
00000004
02F61F03
40200000
$ sudo cat /sys/class/mei/mei0/dev_state
ENABLED% 
```

😢