thesofproject/sof #10034: [BUG] No sound from internal speakers

estnml, 2025-05-27:
**Describe the bug**
I'm experiencing a persistent audio issue with my Arch Linux setup and am hoping to get some guidance. My internal speakers do not produce any sound. This happens after when i power up my machine, or after suspend-to-ram. To get them working, I consistently have to run a series of hda-verb commands in the terminal (found them in web after some research, they are working but i dont know what they do exatly). This is the [link](https://bugzilla.kernel.org/show_bug.cgi?id=207423#c94) of the thread that i found the scripts, although i had to change the card and channel number.

**To Reproduce**
None

**Reproduction Rate**
9 out of 10 times after startup or resume from suspend.

**Expected behavior**
Laptop sound working, after startup and resuming from suspend.

**Impact**
annoyance

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: Linux 6.14.7-arch2-1
    * SOF: main v2.12.1 / f46921ffc67107f1c9e67c22307d63d370f18fcb
2) Name of the topology file
    * Topology: intel/sof-tplg/sof-hda-generic-2ch.tplg
3) Name of the platform(s) on which the bug is observed.
device: Razer Blade 18 2023
system info: Linux 6.14.7-arch2-1 SMP PREEMPT_DYNAMIC Thu, 22 May 2025 05:37:49 +0000 x86_64 GNU/Linux
window manager: niri (wayland)
cpu: 13th Gen Intel(R) Core(TM) i9-13950HX (32) @ 5.50 GHz
GPU 1: NVIDIA GeForce RTX 4080 Max-Q / Mobile [Discrete]
GPU 2: Intel Raptor Lake-S UHD Graphics @ 1.65 GHz [Integrated]

**Screenshots or console output**
sudo dmesg | rg "sof-|snd_"
```
[    3.703328] snd_hda_intel 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    4.115733] snd_hda_intel 0000:01:00.1: enabling device (0000 -> 0002)
[    4.115874] snd_hda_intel 0000:01:00.1: Disabling MSI
[    4.115886] snd_hda_intel 0000:01:00.1: Handle vga_switcheroo audio client
[    4.202169] sof-audio-pci-intel-tgl 0000:00:1f.3: enabling device (0000 -> 0002)
[    4.202292] sof-audio-pci-intel-tgl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[    4.202324] sof-audio-pci-intel-tgl 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    4.209293] sof-audio-pci-intel-tgl 0000:00:1f.3: use msi interrupt mode
[    4.230601] sof-audio-pci-intel-tgl 0000:00:1f.3: hda codecs found, mask 5
[    4.230603] sof-audio-pci-intel-tgl 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[    4.230605] sof-audio-pci-intel-tgl 0000:00:1f.3: NHLT device BT(0) detected, ssp_mask 0x4
[    4.230606] sof-audio-pci-intel-tgl 0000:00:1f.3: BT link detected in NHLT tables: 0x4
[    4.230607] sof-audio-pci-intel-tgl 0000:00:1f.3: DMICs detected in NHLT tables: 2
[    4.232699] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware paths/files for ipc type 0:
[    4.232701] sof-audio-pci-intel-tgl 0000:00:1f.3:  Firmware file:     intel/sof/sof-rpl-s.ri
[    4.232702] sof-audio-pci-intel-tgl 0000:00:1f.3:  Topology file:     intel/sof-tplg/sof-hda-generic-2ch.tplg
[    4.232928] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 2:2:0-57864
[    4.232929] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:22:1 Kernel ABI 3:23:1
[    4.321138] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware info: version 2:2:0-57864
[    4.321142] sof-audio-pci-intel-tgl 0000:00:1f.3: Firmware: ABI 3:22:1 Kernel ABI 3:23:1
[    4.330651] sof-audio-pci-intel-tgl 0000:00:1f.3: Topology: ABI 3:22:1 Kernel ABI 3:23:1
[    4.353848] snd_hda_codec_realtek ehdaudio0D0: autoconfig for ALC298: line_outs=1 (0x17/0x0/0x0/0x0/0x0) type:speaker
[    4.353850] snd_hda_codec_realtek ehdaudio0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[    4.353851] snd_hda_codec_realtek ehdaudio0D0:    hp_outs=1 (0x21/0x0/0x0/0x0/0x0)
[    4.353852] snd_hda_codec_realtek ehdaudio0D0:    mono: mono_out=0x0
[    4.353852] snd_hda_codec_realtek ehdaudio0D0:    inputs:
[    4.353853] snd_hda_codec_realtek ehdaudio0D0:      Mic=0x18
[    4.409904] input: sof-hda-dsp Mic as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card1/input22
[    4.410005] input: sof-hda-dsp Headphone as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card1/input23
[    4.410030] input: sof-hda-dsp HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card1/input24
[    4.410125] input: sof-hda-dsp HDMI/DP,pcm=4 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card1/input25
[    4.410148] input: sof-hda-dsp HDMI/DP,pcm=5 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card1/input26
```

lsmod | rg sof
```
snd_soc_intel_sof_board_helpers    28672  1 snd_soc_skl_hda_dsp
snd_soc_intel_hda_dsp_common    16384  1 snd_soc_intel_sof_board_helpers
snd_sof_probes         28672  0
snd_sof_pci_intel_tgl    12288  2
snd_sof_pci_intel_cnl    20480  1 snd_sof_pci_intel_tgl
snd_sof_intel_hda_generic    45056  2 snd_sof_pci_intel_tgl,snd_sof_pci_intel_cnl
soundwire_intel        86016  1 snd_sof_intel_hda_generic
snd_sof_intel_hda_common   204800  3 snd_sof_intel_hda_generic,snd_sof_pci_intel_tgl,snd_sof_pci_intel_cnl
snd_soc_hdac_hda       28672  1 snd_sof_intel_hda_common
snd_sof_intel_hda_mlink    36864  3 soundwire_intel,snd_sof_intel_hda_common,snd_sof_intel_hda_generic
snd_sof_intel_hda      20480  2 snd_sof_intel_hda_common,snd_sof_intel_hda_generic
snd_sof_pci            24576  3 snd_sof_intel_hda_generic,snd_sof_pci_intel_tgl,snd_sof_pci_intel_cnl
snd_sof_xtensa_dsp     16384  1 snd_sof_intel_hda_generic
snd_sof               466944  6 snd_sof_pci,snd_sof_intel_hda_common,snd_sof_intel_hda_generic,snd_sof_probes,snd_sof_intel_hda,snd_sof_pci_intel_cnl
snd_sof_utils          16384  1 snd_sof
snd_soc_acpi_intel_match   131072  4 snd_soc_intel_sof_board_helpers,snd_sof_intel_hda_generic,snd_sof_pci_intel_tgl,snd_sof_pci_intel_cnl
snd_soc_acpi           16384  2 snd_soc_acpi_intel_match,snd_sof_intel_hda_generic
snd_hda_ext_core       36864  6 snd_soc_avs,snd_soc_hda_codec,snd_sof_intel_hda_common,snd_soc_hdac_hda,snd_sof_intel_hda_mlink,snd_sof_intel_hda
snd_soc_core          446464  10 snd_soc_avs,snd_soc_hda_codec,soundwire_intel,snd_sof,snd_soc_intel_sof_board_helpers,snd_sof_intel_hda_common,snd_soc_hdac_hda,snd_sof_probes,snd_soc_dmic,snd_soc_skl_hda_dsp
snd_compress           28672  3 snd_soc_avs,snd_soc_core,snd_sof_probes
snd_intel_dspcfg       40960  5 snd_soc_avs,snd_hda_intel,snd_sof,snd_sof_intel_hda_common,snd_sof_intel_hda_generic
snd_intel_sdw_acpi     16384  2 snd_intel_dspcfg,snd_sof_intel_hda_generic
snd_hda_codec         217088  10 snd_hda_codec_generic,snd_soc_avs,snd_hda_codec_hdmi,snd_soc_hda_codec,snd_hda_intel,snd_hda_codec_realtek,snd_soc_intel_hda_dsp_common,snd_soc_hdac_hda,snd_sof_intel_hda,snd_soc_skl_hda_dsp
snd_hda_core          143360  12 snd_hda_codec_generic,snd_soc_avs,snd_hda_codec_hdmi,snd_soc_hda_codec,snd_hda_intel,snd_hda_ext_core,snd_hda_codec,snd_hda_codec_realtek,snd_soc_intel_hda_dsp_common,snd_sof_intel_hda_common,snd_soc_hdac_hda,snd_sof_intel_hda
snd_pcm               200704  14 snd_soc_avs,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,soundwire_intel,snd_sof,snd_sof_intel_hda_common,snd_compress,snd_sof_intel_hda_generic,snd_soc_core,snd_sof_utils,snd_hda_core,snd_pcm_dmaengine
snd                   155648  27 snd_ctl_led,snd_hda_codec_generic,snd_seq,snd_seq_device,snd_hda_codec_hdmi,snd_hwdep,snd_hda_intel,snd_hda_codec,snd_hda_codec_realtek,snd_sof,snd_timer,snd_compress,snd_soc_core,snd_pcm
```

lspci -k | rg Audio -A 5
```
00:1f.3 Multimedia audio controller: Intel Corporation Raptor Lake High Definition Audio Controller (rev 11)
	DeviceName: Onboard - Sound
	Subsystem: Razer USA Ltd. Device 3007
	Kernel driver in use: sof-audio-pci-intel-tgl
	Kernel modules: snd_hda_intel, snd_soc_avs, snd_sof_pci_intel_tgl
00:1f.4 SMBus: Intel Corporation Raptor Lake-S PCH SMBus Controller (rev 11)
--
01:00.1 Audio device: NVIDIA Corporation AD104 High Definition Audio Controller (rev a1)
	Subsystem: Razer USA Ltd. Device 3007
	Kernel driver in use: snd_hda_intel
	Kernel modules: snd_hda_intel
02:00.0 PCI bridge: Intel Corporation Device 1133 (rev 02)
	Subsystem: Razer USA Ltd. Device 3007
```

lgirdwood, 2025-05-28:
adding @shumingfan re missing HDA verbs.

kv2019i, 2025-06-03:
This is a codec driver issue, so moving to kernel side (not a SOF FW issue).