thesofproject/sof #10083: [BUG] HP Omen 14 not detected

pyaggi, 2025-07-01:
The integrated driver is not being detected in the HP OMEN Transcend 14, 

the output of the command:
```
cat /sys/bus/acpi/devices/*/adr | grep 025d  
0x000030025d071101  
0x000330025d131601
```
According to [#4880](https://github.com/thesofproject/linux/issues/4880) the board should be:
```
Firmware file: intel/sof-ipc4/mtl/sof-mtl.ri
Topology file: intel/sof-ace-tplg/sof-mtl-rt711-l0-rt1316-l3-2ch.tplg.gz)
```
but instead it is defaulting to :
```
Firmware file: intel/sof-ipc4/arl/sof-arl.ri  
Topology file: intel/sof-ace-tplg/sof-arl-rt711-l0-2ch.tplg
```

I already checked and the kernel has all the patches described in #4880 applied.
I'm using kernel 6.15.13  and sof-firmware 2025.05 on opensuse

Here is all dmesg output related to sof with debuging information included
```
[    3.397313] [    T126] audit: type=1400 audit(1751281624.332:4): avc:  denied  { read } for  pid=1 comm="systemd" name="sof-dyndbg.conf" dev="nvme0n1p5" ino=504204 scontext=system_u:system_r:init_t:s0 tcontext=unconfined_u:object_r:user_home_t:s0 tclass=file permissive=0
[    3.837045] [    T126] audit: type=1400 audit(1751281624.772:5): avc:  denied  { read } for  pid=753 comm="systemd-modules" name="sof-dyndbg.conf" dev="nvme0n1p5" ino=504204 scontext=system_u:system_r:systemd_modules_load_t:s0 tcontext=unconfined_u:object_r:user_home_t:s0 tclass=file permissive=0
[    4.508059] [    T800] snd_intel_dspcfg:snd_intel_dsp_driver_probe: snd_hda_intel 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    4.508082] [    T800] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: ACPI reports 4 SDW Link devices
[    4.508090] [    T800] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: Link 1 not selected in firmware
[    4.508094] [    T800] snd_hda_intel 0000:00:1f.3: SoundWire enabled on CannonLake+ platform, using SOF driver
[    5.501429] [    T800] snd_hda_intel 0000:01:00.1: enabling device (0000 -> 0002)
[    5.501547] [    T800] snd_hda_intel 0000:01:00.1: Disabling MSI
[    5.501561] [    T800] snd_hda_intel 0000:01:00.1: Handle vga_switcheroo audio client
[    5.595669] [    T800] snd_intel_dspcfg:snd_intel_dsp_driver_probe: sof-audio-pci-intel-mtl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    5.595696] [    T800] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: ACPI reports 4 SDW Link devices
[    5.595705] [    T800] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: Link 1 not selected in firmware
[    5.595710] [    T800] snd_sof_pci:sof_pci_probe: sof-audio-pci-intel-mtl 0000:00:1f.3: PCI DSP detected
[    5.595729] [    T800] sof-audio-pci-intel-mtl 0000:00:1f.3: enabling device (0000 -> 0002)
[    5.595966] [    T800] sof-audio-pci-intel-mtl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[    5.596046] [    T800] sof-audio-pci-intel-mtl 0000:00:1f.3: bound 0000:00:02.0 (ops intel_audio_component_bind_ops [i915])
[    5.602536] [    T800] snd_sof_intel_hda_common:hda_dsp_ctrl_get_caps: sof-audio-pci-intel-mtl 0000:00:1f.3: checking for capabilities at offset 0xc00
[    5.602542] [    T800] snd_sof_intel_hda_common:hda_dsp_ctrl_get_caps: sof-audio-pci-intel-mtl 0000:00:1f.3: found ML capability at 0xc00
[    5.602544] [    T800] snd_sof_intel_hda_common:hda_dsp_ctrl_get_caps: sof-audio-pci-intel-mtl 0000:00:1f.3: checking for capabilities at offset 0x500
[    5.602547] [    T800] snd_sof_intel_hda_common:hda_dsp_ctrl_get_caps: sof-audio-pci-intel-mtl 0000:00:1f.3: found DRSM capability at 0x500
[    5.602549] [    T800] snd_sof_intel_hda_common:hda_dsp_ctrl_get_caps: sof-audio-pci-intel-mtl 0000:00:1f.3: checking for capabilities at offset 0x700
[    5.602553] [    T800] snd_sof_intel_hda_common:hda_dsp_ctrl_get_caps: sof-audio-pci-intel-mtl 0000:00:1f.3: found SPIB capability at 0x700
[    5.602555] [    T800] snd_sof_intel_hda_common:hda_dsp_ctrl_get_caps: sof-audio-pci-intel-mtl 0000:00:1f.3: checking for capabilities at offset 0x800
[    5.602559] [    T800] snd_sof_intel_hda_common:hda_dsp_ctrl_get_caps: sof-audio-pci-intel-mtl 0000:00:1f.3: found DSP capability at 0x800
[    5.603534] [     T43] snd_sof_intel_hda_common:hda_dsp_stream_init: sof-audio-pci-intel-mtl 0000:00:1f.3: hda global caps = 0x1009a01
[    5.603540] [     T43] snd_sof_intel_hda_common:hda_dsp_stream_init: sof-audio-pci-intel-mtl 0000:00:1f.3: detected 9 playback and 10 capture streams
[    5.603704] [     T43] sof-audio-pci-intel-mtl 0000:00:1f.3: use msi interrupt mode
[    5.603708] [     T43] snd_sof_intel_hda_generic:hda_dsp_probe: sof-audio-pci-intel-mtl 0000:00:1f.3: using IPC IRQ 201
[    5.604186] [     T43] snd_sof:snd_sof_pci_update_bits_unlocked: sof-audio-pci-intel-mtl 0000:00:1f.3: Debug PCIR: 00000000 at  00000044
[    5.604192] [     T43] snd_sof_intel_hda_generic:hda_init_caps: sof-audio-pci-intel-mtl 0000:00:1f.3: PP capability, will probe DSP later.
[    5.607006] [     T43] snd_sof:snd_sof_pci_update_bits_unlocked: sof-audio-pci-intel-mtl 0000:00:1f.3: Debug PCIR: 00000000 at  00000048
[    5.611214] [     T43] snd_sof_intel_hda:hda_codec_detect_mask: sof-audio-pci-intel-mtl 0000:00:1f.3: codec_mask = 0x4
[    5.611235] [     T43] snd_sof:snd_sof_pci_update_bits_unlocked: sof-audio-pci-intel-mtl 0000:00:1f.3: Debug PCIR: 00000000 at  00000048
[    5.611239] [     T43] snd_sof:snd_sof_pci_update_bits_unlocked: sof-audio-pci-intel-mtl 0000:00:1f.3: Debug PCIW: 00000040 at  00000048
[    5.612767] [     T43] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: ACPI reports 4 SDW Link devices
[    5.612773] [     T43] snd_intel_sdw_acpi:sdw_intel_scan_controller: acpi PRP00001:00: Link 1 not selected in firmware
[    5.613325] [     T43] snd_sof_intel_hda:hda_codec_probe: sof-audio-pci-intel-mtl 0000:00:1f.3: HDA codec #2 probed OK: response: 8086281d
[    5.613998] [     T43] snd_sof_intel_hda:request_codec_module: snd_hda_codec_hdmi ehdaudio0D2: loading codec module: hdaudio:v8086281Dr00100000a01
[    5.616627] [     T43] sof-audio-pci-intel-mtl 0000:00:1f.3: hda codecs found, mask 4
[    5.616630] [     T43] sof-audio-pci-intel-mtl 0000:00:1f.3: NHLT device BT(0) detected, ssp_mask 0x4
[    5.616632] [     T43] sof-audio-pci-intel-mtl 0000:00:1f.3: BT link detected in NHLT tables: 0x4
[    5.616635] [     T43] snd_intel_dspcfg:intel_nhlt_get_dmic_geo: sof-audio-pci-intel-mtl 0000:00:1f.3: found 1 format definitions
[    5.616637] [     T43] snd_intel_dspcfg:intel_nhlt_get_dmic_geo: sof-audio-pci-intel-mtl 0000:00:1f.3: max channels found 2
[    5.616639] [     T43] snd_intel_dspcfg:intel_nhlt_get_dmic_geo: sof-audio-pci-intel-mtl 0000:00:1f.3: Array with 2 dmics
[    5.616640] [     T43] snd_intel_dspcfg:intel_nhlt_get_dmic_geo: sof-audio-pci-intel-mtl 0000:00:1f.3: dmic number 2 max_ch 2
[    5.616642] [     T43] sof-audio-pci-intel-mtl 0000:00:1f.3: DMICs detected in NHLT tables: 2
[    5.619001] [    T802] snd_soc_rt1316_sdw:rt1316_sdw_init: rt1316-sdca sdw:0:3:025d:1316:01: rt1316_sdw_init
[    5.619005] [    T802] snd_soc_rt1316_sdw:rt1316_read_prop: rt1316-sdca sdw:0:3:025d:1316:01: rt1316_read_prop
[    5.619655] [     T43] snd_sof:sof_test_topology_file: sof-audio-pci-intel-mtl 0000:00:1f.3: Failed to open topology file: intel/sof-ace-tplg/sof-arl-rt711-l0-2ch.tplg
[    5.619659] [     T43] sof-audio-pci-intel-mtl 0000:00:1f.3: SOF firmware and/or topology file not found.
[    5.619663] [     T43] sof-audio-pci-intel-mtl 0000:00:1f.3: Supported default profiles
[    5.619664] [     T43] sof-audio-pci-intel-mtl 0000:00:1f.3: - ipc type 1 (Requested):
[    5.619666] [     T43] sof-audio-pci-intel-mtl 0000:00:1f.3:  Firmware file: intel/sof-ipc4/arl/sof-arl.ri
[    5.619667] [     T43] sof-audio-pci-intel-mtl 0000:00:1f.3:  Topology file: intel/sof-ace-tplg/sof-arl-rt711-l0-2ch.tplg
[    5.619669] [     T43] sof-audio-pci-intel-mtl 0000:00:1f.3: Check if you have 'sof-firmware' package installed.
[    5.619670] [     T43] sof-audio-pci-intel-mtl 0000:00:1f.3: Optionally it can be manually downloaded from:
[    5.619672] [     T43] sof-audio-pci-intel-mtl 0000:00:1f.3:    https://github.com/thesofproject/sof-bin/
[    5.620189] [     T43] snd_sof_pci_intel_mtl:mtl_enable_sdw_irq: sof-audio-pci-intel-mtl 0000:00:1f.3: FW Poll Status: reg[0x114c]=0x0 successful
[    5.621269] [     T43] snd_sof_pci_intel_mtl:mtl_power_down_dsp: sof-audio-pci-intel-mtl 0000:00:1f.3: FW Poll Status: reg[0x1000]=0x0 successful
[    5.621514] [     T43] sof-audio-pci-intel-mtl 0000:00:1f.3: error: sof_probe_work failed err: -2

```

(by error I posted this issue in the thsofproject/linux , I think here is the right place )

kv2019i, 2025-07-01:
@bardliao Can you take a look? Why the 1316 amps are not found and kernel tries to look for tplg with just the 711 codec?

@pyaggi Can you attach output of alsa-info (see https://thesofproject.github.io/latest/getting_started/intel_debug/suggestions.html) so we can verify the machine configuration?

ujfalusi, 2025-07-01:
Duplicate (same issue) of https://github.com/thesofproject/linux/issues/5470 ?

bardliao, 2025-07-01:
@pyaggi It is likely that below config is not listed in soc-acpi-intel-arl-match.c.
```
0x000030025d071101  
0x000330025d131601
```
The https://github.com/thesofproject/linux/commit/0f8edb15fb6e436f0da7ab25ffcbcaab3def7e8c commit is for MTL and you need a similar commit on soc-acpi-intel-arl-match.c.
In fact, thsofproject/linux  is a better place to report this issue. Let's move to thsofproject/linux and close ttps://github.com/thesofproject/linux/issues/5470