thesofproject/sof #10087: Adding support for two Lenovo Laptops (P1 & P16)

simontrimmer, 2025-07-02:
(Moving from https://github.com/thesofproject/linux/issues/5480)

@bardliao looped us into a conversation - there are a few Lenovo products coming that require quirking in sof_sdw.c and some new topology files.

An issue was the ordering of the arl match table ( https://patch.msgid.link/20250626141841.77780-1-ckeepax@opensource.cirrus.com ), I've just sent a patch to Mark to add a few entries to the sof_sdw_ssid_quirk_table[] to activate the SOC_SDW_CODEC_MIC on these SSIDs ( https://lore.kernel.org/linux-sound/20250702112235.377479-1-simont@opensource.cirrus.com/ )

That done these products then look to load topologies that are new, here are the gathered IDs:

sof-arl-cs42l43-l0-cs35l56-l2-2ch.tplg
[ 9.719802] sof_sdw sof_sdw: create dai link SDW0-Playback-SimpleJack, id 0
[ 9.719805] sof_sdw sof_sdw: create dai link SDW0-Capture-SimpleJack, id 1
[ 9.719807] sof_sdw sof_sdw: create dai link SDW2-Playback-SmartAmp, id 2
[ 9.719808] sof_sdw sof_sdw: create dai link dmic01, id 3
[ 9.719809] sof_sdw sof_sdw: create dai link dmic16k, id 4
[ 9.719810] sof_sdw sof_sdw: create dai link iDisp1, id 5
[ 9.719811] sof_sdw sof_sdw: create dai link iDisp2, id 6
[ 9.719812] sof_sdw sof_sdw: create dai link iDisp3, id 7

sof-arl-cs42l43-l2-cs35l56-l3-2ch.tplg
[ 5.534309] sof_sdw sof_sdw: create dai link SDW2-Playback-SimpleJack, id 0
[ 5.534311] sof_sdw sof_sdw: create dai link SDW2-Capture-SimpleJack, id 1
[ 5.534313] sof_sdw sof_sdw: create dai link SDW3-Playback-SmartAmp, id 2
[ 5.534314] sof_sdw sof_sdw: create dai link dmic01, id 3
[ 5.534314] sof_sdw sof_sdw: create dai link dmic16k, id 4
[ 5.534315] sof_sdw sof_sdw: create dai link iDisp1, id 5
[ 5.534316] sof_sdw sof_sdw: create dai link iDisp2, id 6
[ 5.534317] sof_sdw sof_sdw: create dai link iDisp3, id 7

I made a patch to build them - but testing feedback is that the recording level of the dmic is low. I'm wondering if that is related to building the topology from an old branch (v2.11.1) or whether more cmake magic is required, I know @singalsu was doing some cmake magic recently so wonder if he had any suggestions?

I'll make a PR with the code in it, but the entries I added to tools/topology/topology2/production/tplg-targets-ace1.cmake are (and attached):

"cavs-sdw;sof-arl-cs42l43-l0-cs35l56-l2-2ch;PLATFORM=mtl,NUM_SDW_AMP_LINKS=1,
SDW_AMP_FEEDBACK=false,SDW_SPK_STREAM=Playback-SmartAmp,
SDW_JACK_OUT_STREAM=Playback-SimpleJack,SDW_JACK_IN_STREAM=Capture-SimpleJack,
NUM_DMICS=2,PDM1_MIC_A_ENABLE=0,PDM1_MIC_B_ENABLE=0,DMIC0_ID=3,DMIC1_ID=4,
HDMI1_ID=5,HDMI2_ID=6,HDMI3_ID=7"

"cavs-sdw;sof-arl-cs42l43-l2-cs35l56-l3-2ch;PLATFORM=mtl,NUM_SDW_AMP_LINKS=1,
SDW_AMP_FEEDBACK=false,SDW_SPK_STREAM=Playback-SmartAmp,
SDW_JACK_OUT_STREAM=Playback-SimpleJack,SDW_JACK_IN_STREAM=Capture-SimpleJack,
NUM_DMICS=2,PDM1_MIC_A_ENABLE=0,PDM1_MIC_B_ENABLE=0,DMIC0_ID=3,DMIC1_ID=4,
HDMI1_ID=5,HDMI2_ID=6,HDMI3_ID=7"

[0001-topology2-Add-sof-arl-cs42l43-l2-cs35l56-l3-2ch-supp.patch](https://github.com/user-attachments/files/21017371/0001-topology2-Add-sof-arl-cs42l43-l2-cs35l56-l3-2ch-supp.patch)
[0002-topology2-Add-sof-arl-cs42l43-l0-cs35l56-l2-2ch-supp.patch](https://github.com/user-attachments/files/21017370/0002-topology2-Add-sof-arl-cs42l43-l0-cs35l56-l2-2ch-supp.patch)

simontrimmer, 2025-07-16:
Good!

FYI - I sent a top-of-tree build of these topology files with these suggested changes for testing on target yesterday

bardliao, 2025-07-18:
> Feedback was good - updated the PR !

@simontrimmer Sorry but which PR are you referring to?

singalsu, 2025-07-21:
Thanks @bardliao @simontrimmer ! Yep, the patch add to enable DRC and beamformer looks good. It enables the capture pre-preprocessing with a generic setup for the algorithms.

@simontrimmer It's also possible to use bespoke tuning for capture and render processing (SOF equalizers, DRC, beamformer) via UCM2 DMI ID based scripts, see https://github.com/alsa-project/alsa-ucm-conf/tree/master/ucm2/blobs/sof and https://github.com/alsa-project/alsa-ucm-conf/blob/master/ucm2/Intel/sof-hda-dsp/dsp.conf . Though the example is only for a very simple UP2 board. The UCM variables to set names vary a bit for HDA and SDW topologies with/without DMIC.

closed 2025-07-22 by lgirdwood