thesofproject/sof #10095: [BUG] No speaker or headphone output on ASUS TP3607S (Flip 16) Laptop

WhyKickAmooCow, 2025-07-03:
**Describe the bug**
Under Linux, headphones are not detected, and though a speaker device is present, no audio played through it is output.

Opening a new issue also described in https://github.com/thesofproject/sof/issues/10059#issuecomment-2986099983 as advised by https://github.com/thesofproject/sof/issues/10059#issuecomment-2987323131.

**To Reproduce**
Attempt to play sound through headphone or speaker output on a TP3607S Laptop using Fedora 42.

**Reproduction Rate**
100%

**Expected behavior**
Sound to be output using the selected device.

**Environment**
Fedora 42 - Kernel 6.15.4-200.fc42.x86_64

**Screenshots or console output**
<details>
<summary>Output of `alsa-info.sh --with-devices`:</summary>
```
upload=true&script=true&cardinfo=
!!################################
!!ALSA Information Script v 0.5.3
!!################################

!!Script ran on: Thu Jul  3 21:54:45 UTC 2025

!!Linux Distribution
!!------------------

Fedora release 42 (Adams) NAME="Fedora Linux" ID=fedora PRETTY_NAME="Fedora Linux 42 (KDE Plasma Desktop Edition)" LOGO=fedora-logo-icon CPE_NAME="cpe:/o:fedoraproject:fedora:42" DEFAULT_HOSTNAME="fedora" HOME_URL="https://fedoraproject.org/" DOCUMENTATION_URL="https://docs.fedoraproject.org/en-US/fedora/f42/system-administrators-guide/" SUPPORT_URL="https://ask.fedoraproject.org/" BUG_REPORT_URL="https://bugzilla.redhat.com/" REDHAT_BUGZILLA_PRODUCT="Fedora" REDHAT_BUGZILLA_PRODUCT_VERSION=42 REDHAT_SUPPORT_PRODUCT="Fedora" REDHAT_SUPPORT_PRODUCT_VERSION=42 Fedora release 42 (Adams) Fedora release 42 (Adams)

!!DMI Information
!!---------------

Manufacturer:      ASUSTeK COMPUTER INC.
Product Name:      ASUS Vivobook 16 Flip TP3607SA_TP3607SA
Product Version:   1.0
Firmware Version:  TP3607SA.300
System SKU:        
Board Vendor:      ASUSTeK COMPUTER INC.
Board Name:        TP3607SA

!!ACPI Device Status Information
!!---------------

/sys/bus/acpi/devices/ACPI0003:00/status 	 15
/sys/bus/acpi/devices/ACPI000C:00/status 	 15
/sys/bus/acpi/devices/ACPI000E:00/status 	 15
/sys/bus/acpi/devices/ASCF1201:00/status 	 15
/sys/bus/acpi/devices/ASUH2024:00/status 	 15
/sys/bus/acpi/devices/ASUS2018:00/status 	 15
/sys/bus/acpi/devices/INT33A1:00/status 	 15
/sys/bus/acpi/devices/INT33D3:00/status 	 15
/sys/bus/acpi/devices/INT340E:00/status 	 15
/sys/bus/acpi/devices/INTC105D:00/status 	 15
/sys/bus/acpi/devices/INTC105D:01/status 	 15
/sys/bus/acpi/devices/INTC105D:02/status 	 15
/sys/bus/acpi/devices/INTC105D:03/status 	 15
/sys/bus/acpi/devices/INTC105D:04/status 	 15
/sys/bus/acpi/devices/INTC1068:00/status 	 15
/sys/bus/acpi/devices/INTC1069:01/status 	 15
/sys/bus/acpi/devices/INTC1069:02/status 	 15
/sys/bus/acpi/devices/INTC1069:03/status 	 15
/sys/bus/acpi/devices/INTC1069:04/status 	 15
/sys/bus/acpi/devices/INTC1069:05/status 	 15
/sys/bus/acpi/devices/INTC1069:06/status 	 15
/sys/bus/acpi/devices/INTC1069:07/status 	 15
/sys/bus/acpi/devices/INTC107B:00/status 	 15
[... 778 lines cut ...]
--endcollapse--

!!ALSA Device nodes
!!-----------------

crw-rw----+ 1 root audio 116, 11 Jul  3 19:41 /dev/snd/controlC0
crw-rw----+ 1 root audio 116, 10 Jul  3 19:41 /dev/snd/hwC0D0
crw-rw----+ 1 root audio 116,  9 Jul  3 19:41 /dev/snd/hwC0D2
crw-rw----+ 1 root audio 116,  3 Jul  3 22:07 /dev/snd/pcmC0D0c
crw-rw----+ 1 root audio 116,  2 Jul  4 09:54 /dev/snd/pcmC0D0p
crw-rw----+ 1 root audio 116,  4 Jul  3 22:07 /dev/snd/pcmC0D31p
crw-rw----+ 1 root audio 116,  6 Jul  3 22:07 /dev/snd/pcmC0D3p
crw-rw----+ 1 root audio 116,  7 Jul  3 22:07 /dev/snd/pcmC0D4p
crw-rw----+ 1 root audio 116,  8 Jul  3 22:07 /dev/snd/pcmC0D5p
crw-rw----+ 1 root audio 116,  5 Jul  3 22:07 /dev/snd/pcmC0D6c
crw-rw----+ 1 root audio 116,  1 Jul  3 19:41 /dev/snd/seq
crw-rw----+ 1 root audio 116, 33 Jul  3 19:41 /dev/snd/timer

/dev/snd/by-path:
total 0
drwxr-xr-x. 2 root root  60 Jul  4 09:54 .
drwxr-xr-x. 3 root root 300 Jul  4 09:54 ..
lrwxrwxrwx. 1 root root  12 Jul  3 19:41 pci-0000:00:1f.3-platform-skl_hda_dsp_generic -> ../controlC0
```
</details>

WhyKickAmooCow, 2025-07-05:
Some additional info from dmesg:
```
[    6.112599] tas2781-hda i2c-TIAS2781:00: Failed to get Speaker id.
[    6.112616] tas2781-hda i2c-TIAS2781:00: read acpi error, ret: -2
[    6.112621] tas2781-hda i2c-TIAS2781:00: error -ENOENT: Platform not supported
[    6.112844] tas2781-hda i2c-TIAS2781:00: probe with driver tas2781-hda failed with error -2
[    6.394659] sof-audio-pci-intel-lnl 0000:00:1f.3: enabling device (0000 -> 0002)
[    6.394865] sof-audio-pci-intel-lnl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[    6.394939] sof-audio-pci-intel-lnl 0000:00:1f.3: bound 0000:00:02.0 (ops intel_audio_component_bind_ops [xe])
[    6.401962] sof-audio-pci-intel-lnl 0000:00:1f.3: use msi interrupt mode
[    6.456937] sof-audio-pci-intel-lnl 0000:00:1f.3: hda codecs found, mask 5
[    6.456942] sof-audio-pci-intel-lnl 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[    6.456944] sof-audio-pci-intel-lnl 0000:00:1f.3: NHLT device BT(0) detected, ssp_mask 0x4
[    6.456945] sof-audio-pci-intel-lnl 0000:00:1f.3: BT link detected in NHLT tables: 0x4
[    6.456946] sof-audio-pci-intel-lnl 0000:00:1f.3: DMICs detected in NHLT tables: 2
[    6.495678] sof-audio-pci-intel-lnl 0000:00:1f.3: Firmware paths/files for ipc type 1:
[    6.495682] sof-audio-pci-intel-lnl 0000:00:1f.3:  Firmware file:     intel/sof-ipc4/lnl/sof-lnl.ri
[    6.495683] sof-audio-pci-intel-lnl 0000:00:1f.3:  Firmware lib path: intel/sof-ipc4-lib/lnl
[    6.495683] sof-audio-pci-intel-lnl 0000:00:1f.3:  Topology file:     intel/sof-ipc4-tplg/sof-hda-generic-2ch.tplg
[    6.522980] sof-audio-pci-intel-lnl 0000:00:1f.3: Loaded firmware library: ADSPFW, version: 2.13.0.1
[    6.792338] sof-audio-pci-intel-lnl 0000:00:1f.3: Booted firmware version: 2.13.0.1
[    6.847067] sof-audio-pci-intel-lnl 0000:00:1f.3: Topology: ABI 3:29:1 Kernel ABI 3:23:1
[    6.854186] sof-audio-pci-intel-lnl 0000:00:1f.3: Loaded firmware library: ADSPFW, version: 2.13.0.1
[    7.348258] input: sof-hda-dsp Headphone as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card1/input22
[    7.348402] input: sof-hda-dsp HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card1/input23
[    7.348493] input: sof-hda-dsp HDMI/DP,pcm=4 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card1/input24
[    7.348559] input: sof-hda-dsp HDMI/DP,pcm=5 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card1/input25
```

kv2019i, 2025-08-05:
Moving to kernel as this seems to be a codec driver issue (the TI amp driver "probe with driver tas2781-hda failed with error").