thesofproject/sof #10201: [BUG] Audio not working on HP Omen Transcend 14 (2025) with Intel Core Ultra 7 255H (ARL) - driver is requesting wrong topology file.

jkfids, 2025-08-29:
**Description**
The `sof-audio-pci-intel-mtl` driver on the HP Omen Transcend 14 (2025) with the Intel Core Ultra 7 255H (Arrow Lake) processor incorrectly requests a topology file that does not exist. The audio device loads with a "dummy output" because it cannot find the required firmware and topology files, leading to a `-22` component load failure. To fix the issue on my end, I manually downloaded the meteor lake files `sof-mtl.ri` and `sof-tml-rt711-l0-rt1316-l3-2ch.tplg` from `sof-bin` and copied/renamed them to the directory expected by the kernel via the commands:

```
sudo cp /path/to/downloaded/sof-mtl.ri /lib/firmware/intel/sof-ipc4/arl/sof-arl.ri
```

```
sudo cp /path/to/downloaded/sof-mtl-rt711-l0-rt1316-l3-2ch.tplg /lib/firmware/intel/sof-ace-tplg/sof-arl-rt711-l0-rt1316-l3-2ch.tplg
```

**Expected behavior**
The kernel should load the correct firmware for the audio device, and sound should work.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: v6.16
    * SOF: v2.13.x
2) Name of the topology file
    * Topology: The kernel requested `sof-arl-rt711-l0-rt1316-l3-2ch.tplg`, but the working file from the SOF repository was actually `sof-mtl-rt722-l0_rt1320-l2.tplg`.
3) Name of the platform(s) on which the bug is observed.
    * Platform: HP Omen Transcend 14 (2025) with Intel Core Ultra 7 255H (Arrow Lake)
    * OS: Arch, Garuda, Ubuntu 25.04, Fedora 42. Fix verified in Ubuntu and Fedora.

**Screenshots or console output**
Before fix:
[image]

After fix:
[image]

lgirdwood, 2025-08-29:
Looks like this topology is missing in https://github.com/thesofproject/sof-bin/tree/main/v2.13.x/sof-ipc4-tplg-v2.13
@bardliao fyi can the mtl topology above be renamed and reused ?

bardliao, 2025-08-31:
@lgirdwood Yes, and PR https://github.com/thesofproject/sof/pull/10190 added the topologies.

closed 2025-11-03 by lgirdwood

YukkiEats, 2025-11-17:
did you ever fix this? i recently put arch on my omen transend and find i have no audio